nickpourazima/rotary-phone-audio-guestbook #5: Invalid sample rate error when running on boot

krazywake, 2023-08-09:
Hi Nick, first off thanks for the very useful tutorial and script. This is one of my first rpi projects. I've got everything working beautifully except for when audioGuestBook.py tries to automatically run during boot up.

When I check the status I'm getting this in return:

● audioGuestBook.service - Rotary Phone Guest Book Project
     Loaded: loaded (/etc/systemd/system/audioGuestBook.service; enabled; vendo>
     Active: active (running) since Tue 2023-08-08 16:09:28 MST; 7h ago
   Main PID: 441 (python3)
      Tasks: 3 (limit: 414)
        CPU: 2min 1.022s
     CGroup: /system.slice/audioGuestBook.service
             └─441 python3 /home/<user>/rotary-phone-audio-guestbook/audioGuest>

Aug 08 16:11:45 raspberrypi env[441]:     play(
Aug 08 16:11:45 raspberrypi env[441]:   File "/home/<user>/.local/lib/python3.9>
Aug 08 16:11:45 raspberrypi env[441]:     _play_with_pyaudio(audio_segment)
Aug 08 16:11:45 raspberrypi env[441]:   File "/home/<user>/.local/lib/python3.9>
Aug 08 16:11:45 raspberrypi env[441]:     stream = p.open(format=p.get_format_f>
Aug 08 16:11:45 raspberrypi env[441]:   File "/home/<user>/.local/lib/python3.9>
Aug 08 16:11:45 raspberrypi env[441]:     stream = PyAudio.Stream(self, *args, >
Aug 08 16:11:45 raspberrypi env[441]:   File "/home/<user>/.local/lib/python3.9>
Aug 08 16:11:45 raspberrypi env[441]:     self._stream = pa.open(**arguments)
Aug 08 16:11:45 raspberrypi env[441]: OSError: [Errno -9997] Invalid sample rate
lines 1-19/19 (END)

If I navigate the terminal to the project folder and run the script manually, it works just fine. Any thoughts on what's causing this or how to fix it? FYI, I'm not a programmer by any stretch, but I have access to my brother who is very experienced, but figured I'd ask you first since you may have run into this using this same combination of dependencies, etc. Thanks in advance!

nickpourazima, 2023-08-10:
Hi guys, appreciate the feedback!

# Debugging
It seems like you both might be running into a similar issue so I'm curious as to a couple things: 

1. Which rpi are you using?
2. Which usb audio interface?
3. Try modifying the `sample_rate` parameter in the projects `config.yaml` file based on the output of the following script:

```
#!/usr/bin/env python3
import pyaudio

p = pyaudio.PyAudio()

for i in range(p.get_device_count()):
    device_info = p.get_device_info_by_index(i)
    print(f"Device {i}: {device_info['name']} - Sample Rate: {device_info['defaultSampleRate']}")

p.terminate()
```

## To run it:
- copy this script into a new file and call in `test.py` for example
- open a terminal
- `chmod +x test.py` to give it appropriate executable permissions
- `./test`

## Determination
The output should give you information about your usb audio interface device and it's sampling rate capabilities.
My guess is that whatever hardware you're using isn't compatible with a 44.1kHz sample rate (which is the default set in the `config.yaml`) so modify that to whatever necessary. For example, some fancier cards can do 96kHz or even 192kHz but if it's a lower quality one it might only be able to do 16kHz.

nickpourazima, 2023-08-10:
A few other thoughts:

- I assumed you are operating as an admin (sudo privileges) on your rpi, if that's not the case we might need to debug some other things. For example: 
  -  If you installed the python package dependencies local to your username you might need to install them system wide instead since the system service might not have access to those packages.
- Try changing this line in the [audioGuestBook.service](https://github.com/nickpourazima/rotary-phone-audio-guestbook/blob/main/audioGuestBook.service) file to make sure that the script doesn't start until the ALSA service is ready:
`After=multi-user.target alsa-restore.service alsa-state.service`
  -  Note: I haven't tested this out yet, so be aware.

krazywake, 2023-08-14:
Hi Nick, thanks for the detailed reply.

I am using a Pi Zero W with this USB audio adapter (https://a.co/d/bg4fExx). I tried adding your modification to make sure the script doesn't start until ALSA service is ready, but didn't seem to help. I ran your test script with the following output:

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.front.0:CARD=0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.iec958.0:CARD=0,AES0=6,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
[... cut ...]
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_dmix.c:1075:(snd_pcm_dmix_open) unable to open slave
Playing voicemail message...
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
Playing beep...
recording
Finished recording!
Phone on hook.
Sleeping...

Would love any further troubleshooting steps if you've got them. Thanks!

krazywake, 2023-08-15:
When running the pi in "headless" mode instead of the GUI I am seeing this message instead of the Sample Rate error:

`Aug 14 22:28:31 raspberrypi env[421]: OSError: [Errno -9996] Invalid output device (no default output device)`

isaacmg410, 2023-08-15:
Hi @nickpourazima . Thanks for your answer.
I'm using a raspberry pi 3 Model B and this audio interface. https://www.amazon.es/dp/B01N905VOY
This is the result of the test.py file
```
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Device 0: USB Audio Device: - (hw:1,0) - Sample Rate: 44100.0
Device 1: pulse - Sample Rate: 44100.0
Device 2: default - Sample Rate: 44100.0
```

So, I assume that the sample_rate at `44100` is it ok, or should I add the decimal also?
Yes the user, is operating as admin.

after the modifications in the service,
I'm seeing this.

````
isaacmassot@raspberrypi:/etc/systemd/system $ sudo systemctl status audioGuestBook.service
● audioGuestBook.service - Rotary Phone Guest Book Project
     Loaded: loaded (/etc/systemd/system/audioGuestBook.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-08-15 10:35:49 BST; 12s ago
   Main PID: 2961 (python3)
      Tasks: 6 (limit: 1595)
        CPU: 3.483s
     CGroup: /system.slice/audioGuestBook.service
             └─2961 python3 /home/isaacmassot/Desktop/rotaryGuestBook/audioGuestBook.py

Aug 15 10:35:55 raspberrypi env[2961]:     play(
Aug 15 10:35:55 raspberrypi env[2961]:   File "/home/isaacmassot/.local/lib/python3.9/site-packages/pydub/playback.py", line 64, in play
Aug 15 10:35:55 raspberrypi env[2961]:     _play_with_pyaudio(audio_segment)
Aug 15 10:35:55 raspberrypi env[2961]:   File "/home/isaacmassot/.local/lib/python3.9/site-packages/pydub/playback.py", line 23, in _play_with_pyaudio
Aug 15 10:35:55 raspberrypi env[2961]:     stream = p.open(format=p.get_format_from_width(seg.sample_width),
Aug 15 10:35:55 raspberrypi env[2961]:   File "/home/isaacmassot/.local/lib/python3.9/site-packages/pyaudio.py", line 754, in open
Aug 15 10:35:55 raspberrypi env[2961]:     stream = Stream(self, *args, **kwargs)
Aug 15 10:35:55 raspberrypi env[2961]:   File "/home/isaacmassot/.local/lib/python3.9/site-packages/pyaudio.py", line 445, in __init__
Aug 15 10:35:55 raspberrypi env[2961]:     self._stream = pa.open(**arguments)
Aug 15 10:35:55 raspberrypi env[2961]: OSError: [Errno -9997] Invalid sample rate
````

krazywake, 2023-08-15:
@nickpourazima I could be way off base here, but It seems to me like for some reason when running at startup Pyaudio is trying to playback on a different "default device" which might not be defined or is trying to use the HDMI audio output. I have been googling the heck out of this but haven't solved it yet. Again, everything works as intended when I run the script manually from the terminal. I even got rid of the buffer underrun error after I switched to only headless CLI mode.

nickpourazima, 2023-08-15:
Hey @krazywake, I'll try to dig into the logs a bit deeper later but wondering if you might have tried creating an `/etc/asound.conf` to explicitly set your audio interface input as the the proper default to use instead of HDMI? Check out this post for some further detail:

https://askubuntu.com/questions/409860/hdmi-audio-on-default-alsa-device-in-xubuntu

Let me know how it goes!

nickpourazima, 2023-08-15:
Awesome! So to conclude:

## Check the Sound Card Configuration:
Verify the available sound devices using the following command:

```bash
aplay -l
```

_Ensure that your USB audio interface is listed and note the card and device numbers._

## Set the Default Sound Card:
If you want to route audio through your USB audio interface, you'll need to make it the default sound card. 
Edit the ALSA configuration file (usually located at `/etc/asound.conf` or `~/.asoundrc`) and add the following:

```bash
defaults.pcm.card X
defaults.ctl.card X
```
_Replace X with the card number of your USB audio interface obtained from the previous step._

## Restart ALSA
```bash
sudo /etc/init.d/alsa-utils restart
```

nickpourazima, 2023-08-15:
@isaacmg410 

To answer your questions, you shouldn't need the decimal in the config file for the sample rate.

Please try the steps above to confirm that it is defaulting to your proper audio interface and let me know how it goes. If it's still not working, let's open up a new issue to track it. Thanks!

isaacmg410, 2023-08-20:
Hi @nickpourazima that one worked. I've another issue right now... I will open a new issue.
Thanks a lot for your help!