nickpourazima/rotary-phone-audio-guestbook #10: Mic Recording Errors

amcco1, 2023-09-26:
I have been beating my head against this thing for the past few hours, I have had a lot of issues, mostly with installing requirements. I had to install all of the following manually: ffmpeg, python3-scipy, python3-pydub, and python3-yaml, python3-pyaudio.

However I am now so close to getting it to work. Hook is working, as is the speaker, however mic recording is not working. After the beep finishes playing, I get the following errors. (There were a lot of PCM errors too, but I fixed those by commenting out devices in alsa.config)

```
Phone off hook, ready to begin!
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Playing voicemail message...
Playing beep...
recording
Expression 'parameters->channelCount <= maxChans' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1514
Expression 'ValidateParameters( inputParameters, hostApi, StreamDirection_In )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2818
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 223, in _call_when_changed
    super(RPiGPIOPin, self)._call_when_changed()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 128, in _call_when_changed
    super(LocalPiPin, self)._call_when_changed(
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 298, in _call_when_changed
    method(ticks, state)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 182, in _pin_changed
    self._fire_events(ticks, bool(self._state_to_value(state)))
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 405, in _fire_events
    self._fire_deactivated()
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 452, in _fire_deactivated
    super(HoldMixin, self)._fire_deactivated()
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 369, in _fire_deactivated
    self.when_deactivated()
  File "/home/pi/rotaryGuestbook/audioGuestBook.py", line 51, in off_hook
    audio_interface.record()
  File "/home/pi/rotaryGuestbook/audioInterface.py", line 26, in record
    self.stream = self.audio.open(
  File "/usr/lib/python3/dist-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9998] Invalid number of channels
Phone on hook.
Sleeping...
^CTraceback (most recent call last):
  File "/home/pi/rotaryGuestbook/audioGuestBook.py", line 73, in <module>
    main()
  File "/home/pi/rotaryGuestbook/audioGuestBook.py", line 69, in main
    pause()
KeyboardInterrupt
``` 

I have tried installing Jack manually, but that has not made any difference. 

Do you have any suggestions?

nickpourazima, 2023-09-28:
Hi @amcco1,

I don't believe any other dependencies should be required outside of those that are in the requirements.txt, were you not able to just `pip install -r requirements.txt` from the repo root directory?

As far as the other errors go, please walk me through your set up so I can get on the same page. 

- What HW are you using? An rpi zero? 
- What audio interface? 
- Were you starting from a fresh raspbian os base or something else?

Regarding the issues:

1) Everything in this project utilizes ALSA directly so Jack is unnecessary

- Ensure that ALSA is configured correctly and not routing through JACK. You can check your ALSA configuration files or use tools like `alsamixer` to inspect your setup.
- If you have JACK installed and you're sure you don't need it, consider either uninstalling it or ensuring it's not started automatically. This might help avoid any unintentional interactions.

2) Regarding: `Expression 'parameters->channelCount <= maxChans' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1514`
Does you microphone record in stereo? If not, you might setting the channels to one in the `config.yaml`.

3) GPIO errors: make sure you have the right pins numbers associated properly, refer to the [diagram](https://github.com/nickpourazima/rotary-phone-audio-guestbook/blob/main/images/rpi_GPIO.png)

Hopefully that's enough to get to unblocked for now.

amcco1, 2023-09-28:
So I have gotten very close to getting it to work, but still not quite. The problem at the moment is that it still will not record, but in this case it is just skipping the record. It plays the voicemail, beep, terminal says recording, then instantly says finished recording, there is no time at all to actually record a message. I will add a video to show what I mean.

- I am using a Pi 3B+ for testing, though I do plan to just move the SD card to a Pi Zero later. 
- Using [this ](https://www.adafruit.com/product/1475)audio interface from Adafruit.
- Started from a fresh install of Raspbian Lite.

Just to make sure I didn't do something wrong, I just swapped the SD card and went through the setup again. Here's the steps I went through.

1. Installed Rasbian Lite. 
2. Updated and upgraded apt
3. Rebooted
4. Installed git
5. Git cloned your repo
6. Installed pip
7. [Installed requirements using pip](https://imgur.com/a/Rn14t7w)
8. Modified config.yaml, changed hook gpio to pin 17 which is what I'm using, changed audio channels to 1.
9. Set the default audio device in /etc/asound.conf
10. Transferred my voicemail.wav file over.
11. Tried to run the .py script, and it fails. Here is error.
```
Traceback (most recent call last):
  File "/home/pi/rotary-phone-audio-guestbook/audioGuestBook.py", line 3, in <module>
    import audioInterface
  File "/home/pi/rotary-phone-audio-guestbook/audioInterface.py", line 3, in <module>
    import pyaudio
ModuleNotFoundError: No module named 'pyaudio'
``` 

So then I went through the process of installing the requirements manually.

```
sudo apt install python3-pyaudio python3-gpiozero python3-pydub python3-yaml -y
``` 

Now the python script will run, the hook works, the speaker works, but the recording does not work. Same result as before.

Here is the full error output. 
```
pi@piphone:~/rotary-phone-audio-guestbook $ sudo python audioGuestBook.py
Phone off hook, ready to begin!
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.USB-Audio.pcm.hdmi.0:CARD=1,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM hdmi
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.USB-Audio.pcm.hdmi.0:CARD=1,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM hdmi
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Playing voicemail message...
Playing beep...
recording
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 223, in _call_when_changed
    super(RPiGPIOPin, self)._call_when_changed()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 128, in _call_when_changed
    super(LocalPiPin, self)._call_when_changed(
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 298, in _call_when_changed
    method(ticks, state)
[... 42 lines cut ...]
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Playing voicemail message...
Playing beep...
recording
Finished recording!
Phone on hook.
Sleeping...
^CTraceback (most recent call last):
  File "/home/pi/rotary-phone-audio-guestbook/audioGuestBook.py", line 73, in <module>
    main()
  File "/home/pi/rotary-phone-audio-guestbook/audioGuestBook.py", line 69, in main
    pause()
KeyboardInterrupt
``` 
[Here is a video showing it. ](https://imgur.com/a/bhKpRus)
I am still seeing the Jack errors, not sure why, but the recording is still not working..
I have tested the mic using this:
```
arecord -f S16_LE -d 10 -r 44100 --device="hw:1,0" test-mic.wav
```
The mic works, I can playback the test recording using aplay with no issues.

nickpourazima, 2023-09-28:
Hmm, can you run:

`chmod +x audioGuestBook.py `
`chmod +x audioInterface.py `

And try again without `sudo`?

If that doesn't work maybe pyaudio's latest version is defaulting to JACK, you can try to debug a little further from this post:

https://stackoverflow.com/questions/36894315/how-to-select-a-specific-input-device-with-pyaudio

I apologize for the limited assistance as I'm traveling right now (without a laptop 😆), but please keep me posted!

amcco1, 2023-10-02:
I'm stumped and don't know what else to try. chmod doesn't seem to make any difference.

I have the alsa_hw_mapping in config.yaml set to 1

Here is the ouput of aplay -l
```
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi [vc4-hdmi], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
``` 

I even tried changing hw mapping to 0 or 2, neither of them worked either.

nickpourazima, 2023-10-03:
Can you:

1. save a copy of your `config.yaml`
2. `git pull`
3. `git checkout develop`
4. paste back your `config.yaml`
5. `chmod +x installer.sh`
6. `./installer.sh`

And let me know if you have better luck with these new modifications I pushed?

amcco1, 2023-10-03:
The installer fails when I run it, fails on the .service part of it. 
```
pi@piphone:~/rotary-phone-audio-guestbook $ ./installer.sh
Installing dependencies...
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-gpiozero is already the newest version (1.6.2-1).
python3-pip is already the newest version (20.3.4-4+rpt1+deb11u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pydub in /home/pi/.local/lib/python3.9/site-packages (0.25.1)
Requirement already satisfied: pyaudio in /home/pi/.local/lib/python3.9/site-packages (0.2.12)
Requirement already satisfied: PyYAML in /home/pi/.local/lib/python3.9/site-packages (6.0)
sed: can't read audioGuestBook.service.template: No such file or directory
Failed to enable unit: Unit file /etc/systemd/system/audioGuestBook.service is masked.
Failed to start audioGuestBook.service: Unit audioGuestBook.service is masked.
``` 
I tried doing systemctl unmask audioGuestBook.service but it made no difference.

I tried to run audioGuestbBook.py manually and it never started recording, throwing various errors.
```
pi@piphone:~/rotary-phone-audio-guestbook $ sudo python audioGuestBook.py
INFO:__main__:Phone on hook.
Sleeping...
INFO:__main__:Phone off hook, ready to begin!
INFO:__main__:Playing voicemail message...
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
INFO:__main__:Playing beep...
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
INFO:__main__:recording
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Traceback (most recent call last):
  File "/home/pi/rotary-phone-audio-guestbook/audioInterface.py", line 100, in close
    wavefile.writeframes(b"".join(self.frames))
  File "/usr/lib/python3.9/wave.py", line 437, in writeframes
    self.writeframesraw(data)
  File "/usr/lib/python3.9/wave.py", line 426, in writeframesraw
    self._ensure_header_written(len(data))
  File "/usr/lib/python3.9/wave.py", line 467, in _ensure_header_written
    self._write_header(datasize)
  File "/usr/lib/python3.9/wave.py", line 471, in _write_header
    self._file.write(b'RIFF')
AttributeError: 'PosixPath' object has no attribute 'write'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 223, in _call_when_changed
    super(RPiGPIOPin, self)._call_when_changed()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 128, in _call_when_changed
    super(LocalPiPin, self)._call_when_changed(
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 298, in _call_when_changed
    method(ticks, state)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 182, in _pin_changed
    self._fire_events(ticks, bool(self._state_to_value(state)))
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 405, in _fire_events
    self._fire_deactivated()
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 452, in _fire_deactivated
    super(HoldMixin, self)._fire_deactivated()
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 369, in _fire_deactivated
    self.when_deactivated()
  File "/home/pi/rotary-phone-audio-guestbook/audioGuestBook.py", line 85, in <lambda>
    hook.when_released = lambda: off_hook(args)
  File "/home/pi/rotary-phone-audio-guestbook/audioGuestBook.py", line 72, in off_hook
    audio_interface.close(output_file)
  File "/home/pi/rotary-phone-audio-guestbook/audioInterface.py", line 100, in close
    wavefile.writeframes(b"".join(self.frames))
  File "/usr/lib/python3.9/wave.py", line 332, in __exit__
    self.close()
  File "/usr/lib/python3.9/wave.py", line 444, in close
    self._ensure_header_written(0)
  File "/usr/lib/python3.9/wave.py", line 467, in _ensure_header_written
    self._write_header(datasize)
  File "/usr/lib/python3.9/wave.py", line 471, in _write_header
    self._file.write(b'RIFF')
AttributeError: 'PosixPath' object has no attribute 'write'
INFO:__main__:Phone on hook.
Sleeping...
^CTraceback (most recent call last):
  File "/home/pi/rotary-phone-audio-guestbook/audioGuestBook.py", line 90, in <module>
    main()
  File "/home/pi/rotary-phone-audio-guestbook/audioGuestBook.py", line 86, in main
    pause()
KeyboardInterrupt
``` 

Thank you for your effort on this btw, I appreciate it.

nickpourazima, 2023-10-04:
Hey @amcco1

Thanks for testing that out. I unfortunately took my rotary phone apart after the wedding, so I can't even validate what I'm coding right now on the actual platform. However, I do know why you were getting these errors and they should be addressed in the commits I just pushed. Can you try again?

nickpourazima, 2023-10-04:
Do you have a "voicemail.wav"  in the sounds directory?  If not, make sure you have that present.

amcco1, 2023-10-04:
Yes I do. It plays the voicemail and beep with no issues. The only issue is recording.

nickpourazima, 2023-10-04:
Hey @amcco1 what's your hook hardware setup? Can you share a picture? I'm wondering if it's immediately toggling from off-hook/on-hook. Could be indicative of wrong wiring.

Also, let me know if this helps:
`sudo apt install portaudio19-dev`

And lastly, I don't think the Jack warnings are too big of an issue, I do remember seeing similar output on mine.

amcco1, 2023-10-04:
Installing portaudio made no difference.

The hook configuration could be it.... mine is opposite of your, it is closed when on the hook, and open when the phone is off the hook. I have inverted your code at the end of audioGuestBook.py thinking that would be enough, but maybe there is something else that needs changed?
```
    hook.when_pressed = on_hook
    hook.when_released = lambda: off_hook(args)
    pause()
``` 

The picture won't tell you much, as it's impossible to see the terminals for the hook since they're on the bottom and it is riveted to the bottom of the frame. I'm using GPIO 17 on pin 11 and ground on pin 6.
[image]

nickpourazima, 2023-10-05:
Ahh, it could be the lambda you added.  The GpioZero docs mention the proper way to call a function with their Button object:
[image]

amcco1, 2023-10-05:
I did not add the lambda, you already have it in the develop branch. All I did was just switch hook.when_pressed and hook.when_released.

nickpourazima, 2023-10-05:
Oh right, my bad. I'll look into this a bit more later and get back to you.

nickpourazima, 2023-10-08:
Hey @amcco1, just pushed some code changes - removed the reliance on lambda, you should be able to swap these no problem:
```python
    hook.when_pressed = off_hook
    hook.when_released = on_hook
```
Let me know how it goes!

amcco1, 2023-10-09:
Well you changes did make a bit of a difference! There are no Jack errors now. 

However, it still doesn't stop to record, it skips straight to "Finished recording"

```
INFO:__main__:Phone off hook, ready to begin!
INFO:__main__:Playing voicemail message...
INFO:__main__:Playing beep...
INFO:__main__:recording
INFO:__main__:Finished recording!
INFO:__main__:Phone on hook.
Sleeping...
```

nickpourazima, 2023-10-09:
Okay so I think I understand what's going on. Since your hook type is Normally Closed (NC),  you'd want to pull the GPIO pin up, making the default state HIGH, so when the button is pressed (i.e., the hook is down), it goes LOW. I adjusted the logic and README to account for either type - with it set up for NC by default. You should be good to go once you pull my latest and just run it (without having to change anything).

nickpourazima, 2023-10-09:
Awesome!! Super glad it worked out for you and I really appreciate you taking the time to debug this with me! Thanks to your help this project has become much more stable and versatile. Wish you all the best and hope your friend has a great wedding!

P.S. This is my "buy me a coffee link": https://www.buymeacoffee.com/dillpicholas

nickpourazima, 2023-10-09:
Closing this issue out as fixed ✅

closed 2023-10-09 by nickpourazima