nickpourazima/rotary-phone-audio-guestbook #20: Sound is garbled after first recording

stephanlinke87, 2023-12-21:
Hi Nick,

First and foremost, thanks for creating and sharing this! Currently working on a phone and already have the second one in queue :) 

I built this using a Raspberry Pi Zero 2W and it works all as expected. However, when recording the second message, the voicemail message played is garbled and kind of sped up. When trying for the third time, there is no beep.wav played and you cannot record. Did you by any chance come across this during development? I think it might have to do with the load on the Pi as it's getting somewhat slower after the first attempt. I already switched the power brick to 5V 2.5A, but to no avail. Perhaps you have an idea. Interestingly, the audio is not garbled after the first attempt when using aplay in terminal :)

nickpourazima, 2023-12-21:
Hey @stephanlinke87,

That's great to hear!

Regarding your current issue, I'm wondering a few things:
- Which USB audio card are you using? 
- Do you know the supported sample rate?
- **File I/O**: 
  - After the first recording, can you confirm that the `on_hook` registered in the logs and the first file has been written? 
  - Is that file playable?
- **ALSA settings**: can you send me the output from:
```bash
aplay -l
cat /etc/asound.conf
```
- can you open a terminal and run something like htop to see if perhaps you might be running out of resources?

nickpourazima, 2023-12-21:
Also, can you try dropping the `buffer_size` in the `config.yaml` down to something like 2048 or 1024 and re-test?

stephanlinke87, 2023-12-21:
Hi Nick, 

Thanks for the quick response on this :) To answer your questions: 

**USB Soundcard** https://www.amazon.de/40520/dp/B06XP5R449?language=en_GB&currency=EUR
**Supported Sampling Rates** 92kHz/192KHz (ADC/DAC)
**File I/O** Yup, on hook is registered, but there's something weird, it doesn't find the sound card at first pick up, only after the second pick up: 

```
stephan@dgg-guestbook:~ $ lsusb
Bus 001 Device 002: ID 0bda:4809 Realtek Semiconductor Corp. USB Audio
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

stephan@dgg-guestbook:~ $ journalctl -u audioGuestBook -f
-- Journal begins at Thu 2023-12-21 12:42:00 CET. --
Dec 21 12:42:09 dgg-guestbook systemd[1]: Started Rotary Phone Guest Book Project.
Dec 21 12:50:24 dgg-guestbook env[584]: INFO:__main__:Phone off hook, ready to begin!
Dec 21 12:50:24 dgg-guestbook env[584]: INFO:__main__:Playing voicemail message...
Dec 21 12:50:24 dgg-guestbook env[584]: ERROR:__main__:Error playing voicemail.wav. Error: [Errno -9996] Invalid output device (no default output device)
Dec 21 12:50:24 dgg-guestbook env[584]: INFO:__main__:Playing beep...
Dec 21 12:50:24 dgg-guestbook env[584]: ERROR:__main__:Error playing beep.wav. Error: [Errno -9996] Invalid output device (no default output device)
Dec 21 12:50:24 dgg-guestbook env[584]: INFO:__main__:recording
Dec 21 12:50:24 dgg-guestbook systemd[1]: audioGuestBook.service: Main process exited, code=killed, status=11/SEGV
Dec 21 12:50:24 dgg-guestbook systemd[1]: audioGuestBook.service: Failed with result 'signal'.
Dec 21 12:50:24 dgg-guestbook systemd[1]: audioGuestBook.service: Consumed 5.355s CPU time.
Dec 21 12:50:24 dgg-guestbook systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 1.
Dec 21 12:50:24 dgg-guestbook systemd[1]: Stopped Rotary Phone Guest Book Project.
Dec 21 12:50:24 dgg-guestbook systemd[1]: audioGuestBook.service: Consumed 5.355s CPU time.
Dec 21 12:50:24 dgg-guestbook systemd[1]: Started Rotary Phone Guest Book Project.
Dec 21 12:50:28 dgg-guestbook env[925]: INFO:__main__:Phone on hook.
Dec 21 12:50:28 dgg-guestbook env[925]: Sleeping...
Dec 21 12:50:29 dgg-guestbook env[925]: INFO:__main__:Phone off hook, ready to begin!
Dec 21 12:50:29 dgg-guestbook env[925]: INFO:__main__:Playing voicemail message...
Dec 21 12:50:40 dgg-guestbook env[925]: INFO:__main__:Playing beep...
Dec 21 12:50:41 dgg-guestbook env[925]: INFO:__main__:recording
Dec 21 12:50:44 dgg-guestbook env[925]: INFO:__main__:Finished recording!
Dec 21 12:50:44 dgg-guestbook env[925]: INFO:__main__:Phone on hook.
Dec 21 12:50:44 dgg-guestbook env[925]: Sleeping...
Dec 21 12:50:45 dgg-guestbook env[925]: INFO:__main__:Phone off hook, ready to begin!
Dec 21 12:50:45 dgg-guestbook env[925]: INFO:__main__:Playing voicemail message...
Dec 21 12:50:56 dgg-guestbook env[925]: INFO:__main__:Playing beep...
Dec 21 12:50:57 dgg-guestbook env[925]: INFO:__main__:recording
Dec 21 12:51:00 dgg-guestbook env[925]: INFO:__main__:Finished recording!
Dec 21 12:51:00 dgg-guestbook env[925]: INFO:__main__:Phone on hook.
Dec 21 12:51:00 dgg-guestbook env[925]: Sleeping...
Dec 21 12:51:04 dgg-guestbook env[925]: INFO:__main__:Phone off hook, ready to begin!
Dec 21 12:51:05 dgg-guestbook env[925]: INFO:__main__:Playing voicemail message...
Dec 21 12:51:15 dgg-guestbook env[925]: INFO:__main__:Playing beep...
Dec 21 12:51:16 dgg-guestbook env[925]: INFO:__main__:recording
Dec 21 12:51:16 dgg-guestbook env[925]: INFO:__main__:Finished recording!
Dec 21 12:51:16 dgg-guestbook env[925]: INFO:__main__:Phone on hook.
Dec 21 12:51:16 dgg-guestbook env[925]: Sleeping...
```
 
The file itself is recorded and playable perfectly fine, no noise or crackling. After the third attempt, no beep.wav is played anymore as well, even though its logged :( 

**Command Line output from aplay and cat /etc/asound**

```bash
stephan@dgg-guestbook:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 1: Audio [USB Audio], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
stephan@dgg-guestbook:~ $ cat /etc/asound.conf
defaults.pcm.card 1
defaults.pcm.device 0
defaults.ctl.card 1
stephan@dgg-guestbook:~ $
```

Not much going on in htop, only the console feels sluggish/slow after the first pickup, compared to the first one. The configuration as of now is the following:
```yaml
alsa_hw_mapping: 1
beep_reduction: 0
buffer_size: 2048
channels: 1
hook_gpio: 22
playback_reduction: 0
recording_limit: 300
rotary_gpio: 23
rotary_hold_repeat: true
rotary_hold_time: 0.25
sample_rate: 44100
source_file: audioGuestBook.py
format: INT16
hook_type: NC # or 'NC'
```

nickpourazima, 2023-12-21:
Can you checkout the [dev commit I just pushed](e787bde4523a2f580fca5e643059138c46116933) and let me know how it runs with your set up? I added further logging and safety checks. If it doesn't address the core issue you were facing would you mind pasting the logs?

If running via the systemctl service:
```bash
journalctl -u audioGuestBook.service >> audioGuestbook_logs.txt
```

Thanks!

stephanlinke87, 2023-12-21:
Awesome, I‘ll check it out tomorrow :) Thanks!

stephanlinke87, 2023-12-22:
Unfortunately, it's the same with the new version, however, after the first pickup, it complains about the sound card missing. The sound comes out garbled at random now, not just for the first attempt :D The load is normal as well, it goes to ~20% on pick up, but quickly goes down again. 
[audioGuestbook_logs.txt](https://github.com/nickpourazima/rotary-phone-audio-guestbook/files/13749455/audioGuestbook_logs.txt)

Note that the last beep in the log didn't play, altough it was logged :s

stephanlinke87, 2023-12-22:
Alright, commenced some further testing. The issue is indeed appearing randomly, but often :D New log, directly after reboot, voicemail.wav sounds garbled/sped up, and it's not recording unfortunately (even though the proper directories exist within the src directory). Here's the log :) 

```
stephan@dgg-guestbook:~/AudioGuestbook $ journalctl -u audioGuestBook -f
-- Journal begins at Fri 2023-12-22 09:49:11 CET. --
Dec 22 09:51:37 dgg-guestbook systemd[1]: Started Rotary Phone Guest Book Project.
Dec 22 09:51:45 dgg-guestbook env[585]: INFO:__main__:Remember to monitor system resources during recording.
Dec 22 09:56:12 dgg-guestbook env[585]: INFO:__main__:Phone off hook, ready to begin!
Dec 22 09:56:12 dgg-guestbook env[585]: INFO:audioInterface:Initializing Audio Interface with sample rate: 44100, format: 8
Dec 22 09:56:13 dgg-guestbook env[585]: Cannot connect to server socket err = No such file or directory
Dec 22 09:56:13 dgg-guestbook env[585]: Cannot connect to server request channel
Dec 22 09:56:13 dgg-guestbook env[585]: jack server is not running or cannot be started
Dec 22 09:56:13 dgg-guestbook env[585]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 22 09:56:13 dgg-guestbook env[585]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 22 09:56:13 dgg-guestbook env[585]: Cannot connect to server socket err = No such file or directory
Dec 22 09:56:13 dgg-guestbook env[585]: Cannot connect to server request channel
Dec 22 09:56:13 dgg-guestbook env[585]: jack server is not running or cannot be started
Dec 22 09:56:13 dgg-guestbook env[585]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 22 09:56:13 dgg-guestbook env[585]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 22 09:56:13 dgg-guestbook env[585]: ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
Dec 22 09:56:13 dgg-guestbook env[585]: ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
Dec 22 09:56:13 dgg-guestbook env[585]: ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection refused
Dec 22 09:56:13 dgg-guestbook env[585]: ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection refused
Dec 22 09:56:13 dgg-guestbook env[585]: ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
Dec 22 09:56:13 dgg-guestbook env[585]: ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
Dec 22 09:56:13 dgg-guestbook env[585]: ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
Dec 22 09:56:13 dgg-guestbook env[585]: Cannot connect to server socket err = No such file or directory
Dec 22 09:56:13 dgg-guestbook env[585]: Cannot connect to server request channel
Dec 22 09:56:13 dgg-guestbook env[585]: jack server is not running or cannot be started
Dec 22 09:56:13 dgg-guestbook env[585]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 22 09:56:13 dgg-guestbook env[585]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 22 09:56:13 dgg-guestbook env[585]: INFO:audioInterface:Audio resources initialized.
Dec 22 09:56:13 dgg-guestbook env[585]: INFO:__main__:Playing voicemail message...
Dec 22 09:56:24 dgg-guestbook env[585]: INFO:__main__:Playing beep...
Dec 22 09:56:25 dgg-guestbook env[585]: INFO:__main__:Recording
Dec 22 09:56:25 dgg-guestbook env[585]: Cannot connect to server socket err = No such file or directory
Dec 22 09:56:25 dgg-guestbook env[585]: Cannot connect to server request channel
Dec 22 09:56:25 dgg-guestbook env[585]: jack server is not running or cannot be started
Dec 22 09:56:25 dgg-guestbook env[585]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 22 09:56:25 dgg-guestbook env[585]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 22 09:56:25 dgg-guestbook env[585]: Cannot connect to server socket err = No such file or directory
Dec 22 09:56:25 dgg-guestbook env[585]: Cannot connect to server request channel
Dec 22 09:56:25 dgg-guestbook env[585]: jack server is not running or cannot be started
Dec 22 09:56:25 dgg-guestbook env[585]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 22 09:56:25 dgg-guestbook env[585]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 22 09:56:25 dgg-guestbook env[585]: ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
Dec 22 09:56:25 dgg-guestbook env[585]: ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
Dec 22 09:56:25 dgg-guestbook env[585]: ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection refused
Dec 22 09:56:25 dgg-guestbook env[585]: ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection refused
Dec 22 09:56:25 dgg-guestbook env[585]: ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
Dec 22 09:56:26 dgg-guestbook env[585]: ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
Dec 22 09:56:26 dgg-guestbook env[585]: ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
Dec 22 09:56:26 dgg-guestbook env[585]: Cannot connect to server socket err = No such file or directory
[... cut ...]
Dec 22 09:56:26 dgg-guestbook env[585]:   File "/home/stephan/AudioGuestbook/src/audioInterface.py", line 206, in stop
Dec 22 09:56:26 dgg-guestbook env[585]:     self.stream.stop_stream()
Dec 22 09:56:26 dgg-guestbook env[585]:   File "/root/.local/lib/python3.9/site-packages/pyaudio/__init__.py", line 500, in stop_stream
Dec 22 09:56:26 dgg-guestbook env[585]:     pa.stop_stream(self._stream)
Dec 22 09:56:26 dgg-guestbook env[585]: OSError: Stream not open
Dec 22 09:56:33 dgg-guestbook env[585]: INFO:__main__:Phone on hook.
Dec 22 09:56:33 dgg-guestbook env[585]: Sleeping...```

nickpourazima, 2023-12-22:
Hmm... 🤔 I will need to look a bit deeper into this. I'm traveling for the Holidays but I'll try to keep you posted. In the meantime, it might be good to rule out SD card corruption. I suspect this could have happened if you had a power supply that didn't meet the rpi zeros needs prior to the new one you purchased.
 
If you have a Linux based machine, can you eject the SD card from the rpi (perhaps make a backup of the contents) and connect it to your computer. Then run something like:
   ```bash
   sudo fsck /dev/sdX
   ```

nickpourazima, 2023-12-22:
One other thing I noticed in the logs was the stated sample rate of 44.1kHz and a bit depth of 8 (which usually sounds pretty crappy). This is way lower than what your USB audio interface supports:

> ADC sampling rate supports max for 96KHz/16bit and DAC sampling rate supports max for 192KHz/24bit.

Try setting the sample rate and bit depth in your ALSA configuration, update the `/etc/asound.conf` file as follows:

```plaintext
defaults.pcm.card 1
defaults.pcm.device 0
defaults.ctl.card 1

pcm.!default {
    type plug
    slave {
        pcm "hw:1,0"
        format S24_3LE # Set to 24-bit depth, adjust as needed
        rate 192000 # Set to 192kHz, adjust as needed
    }
}
```

stephanlinke87, 2023-12-22:
That was it, the audio is working flawless all the time now, even though I had to set the rate to 44100 :) Great work, dude! :) Now it's just for the recording that doesn't ... record :) But perhaps it's working with the current stable branch as well now? Anyways, enjoy your well deserved holidays. Festive greetings from Germany!

nickpourazima, 2023-12-23:
Awesome, glad that fixed it! Yeah, try that config out in the main branch. I haven't even tested the dev branch yet. Hope you have a good holiday as well!

stephanlinke87, 2023-12-25:
So, just a quick update, as I needed to do some further tinkering to alleviate some issues :) 

**Logging issues**
There were some exceptions in the log when its run as a service, which could be alleviated by adding the **sounddevice** module:
```
stephan@dgg-guestbook:~ $ journalctl -u audioGuestBook -f
-- Journal begins at Sun 2023-12-24 15:44:41 CET. --
Dec 24 15:44:51 dgg-guestbook systemd[1]: Started Rotary Phone Guest Book Project.
Dec 24 15:44:58 dgg-guestbook env[584]: INFO:__main__:Remember to monitor system resources during recording.
Dec 24 15:46:12 dgg-guestbook env[584]: INFO:__main__:Phone off hook, ready to begin!
Dec 24 15:46:12 dgg-guestbook env[584]: INFO:audioInterface:Initializing Audio Interface with sample rate: 44100, format: 8
Dec 24 15:46:12 dgg-guestbook env[584]: Cannot connect to server socket err = No such file or directory
Dec 24 15:46:12 dgg-guestbook env[584]: Cannot connect to server request channel
Dec 24 15:46:12 dgg-guestbook env[584]: jack server is not running or cannot be started
Dec 24 15:46:12 dgg-guestbook env[584]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 24 15:46:12 dgg-guestbook env[584]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 24 15:46:12 dgg-guestbook env[584]: Cannot connect to server socket err = No such file or directory
Dec 24 15:46:12 dgg-guestbook env[584]: Cannot connect to server request channel
Dec 24 15:46:12 dgg-guestbook env[584]: jack server is not running or cannot be started
Dec 24 15:46:12 dgg-guestbook env[584]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 24 15:46:12 dgg-guestbook env[584]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec 24 15:46:12 dgg-guestbook env[584]: ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
Dec 24 15:46:12 dgg-guestbook env[584]: ALSA lib pcm_oss.c:377:
```

After adding the sounddevice, there were no longer the exceptions above visible, but new ones when recording. This lead to the recording failing after the third attempt. Couldn't get any audio to play further:
```
lib pcm.c:8545:(snd_pcm_recover) underrun occurred
```
After some googling, someone in the [arch linux forums](https://bbs.archlinux.org/viewtopic.php?id=185736) wrote that configuring the pulse daemon accordingly fixed their issue:
```
default-fragments = 5
default-fragment-size-msec = 2
```
After rebooting, everything works indeed as intended, I did multiple test recordings and it went through smoothly. Not sure if its still required, but I also set the exception handler to false when recording so there's no interruption:
```python
# loop through stream and append audio chunks to frame array
        try:
            start = time.time()
            while self.off_hook_condition():
                if time.time() - start < self.recording_limit:
                    data = self.stream.read(self.chunk, exception_on_overflow=False)
                    self.frames.append(data)
                else:
                    # Notify the user that their recording time is up
                    self.play("time_exceeded.wav")
                    break
```
If this all works for the other phone as well (ordered a different sound card that is smaller), I'll write everything together :)

nickpourazima, 2024-01-17:
Hey @stephanlinke87, hope you had a good holiday!
Great catch there with setting the exception handler to false, I believe I had that set for the deployed version and forgot to change it back in the codebase. Thanks for deep diving into this, I'm glad you were able to figure it out! 

Anyways, I appreciate your detailed response. Let me know if you'd like to submit a push request otherwise I can integrate your changes into my dev branch and we'll go from there.

stephanlinke87, 2024-01-27:
If I push anything, it breaks most of the time :D Sorry, I haven't had the time in the last few weeks :) I'll have the phone back tomorrow from friends and then I'll write down the steps/packages installed. It'll use Ubuntu Server instead (to keep the footprint minimal on the Rasperry Pi Zero 2W). Additionally, the phone also now has a web interface and incorporates RaspAP so the following can be done without much troubles: 

1. Listen to recorded messages
2. Set every message recorded as voicemail
3. Download recorded messages

This might be out of scope for the project, but perhaps we could put this up in the wiki :)

nickpourazima, 2024-02-18:
Those updates sounds pretty sweet @stephanlinke87, up to you if you'd rather keep it as a fork or maybe push the branch with a special ubuntu-server tag or something. Feel free to close this out whenever you're ready!

closed 2024-02-18 by nickpourazima