nickpourazima/rotary-phone-audio-guestbook #22: Something not working

ipanggggg, 2024-02-08:
Hi Nick!

Hope you are well, Thank You for super great project. Im new user for Raspberry pi. Im using Raspberry pi zero 2w to try with your repo. But here im facing some problem where this repo still not running. Here is my log 

`admin@pi:~ $ git clone https://github.com/nickpourazima/rotary-phone-audio-guest                                                                                        book.git
cd rotary-phone-audio-guestbook
Cloning into 'rotary-phone-audio-guestbook'...
remote: Enumerating objects: 115, done.
remote: Counting objects: 100% (70/70), done.
remote: Compressing objects: 100% (54/54), done.
remote: Total 115 (delta 27), reused 38 (delta 10), pack-reused 45
Receiving objects: 100% (115/115), 16.10 MiB | 2.81 MiB/s, done.
Resolving deltas: 100% (32/32), done.
admin@pi:~/rotary-phone-audio-guestbook $ chmod +x installer.sh
./installer.sh
Installing dependencies...
Reading package lists... Done
E: Could not get lock /var/lib/apt/lists/lock. It is held by process 1111 (packagekitd)
N: Be aware that removing the lock file is not a solution and may break your system.
E: Unable to lock directory /var/lib/apt/lists/
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-gpiozero is already the newest version (1.6.2-1).
python3-pip is already the newest version (20.3.4-4+rpt1+deb11u1).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pydub
  Downloading https://www.piwheels.org/simple/pydub/pydub-0.25.1-py2.py3-none-any.whl (32 kB)
Collecting pyaudio
  Downloading https://www.piwheels.org/simple/pyaudio/PyAudio-0.2.14-cp39-cp39-linux_armv7l.whl (56 kB)
     |████████████████████████████████| 56 kB 37 kB/s
Collecting PyYAML
  Downloading https://www.piwheels.org/simple/pyyaml/PyYAML-6.0.1-cp39-cp39-linux_armv7l.whl (45 kB)
     |████████████████████████████████| 45 kB 46 kB/s
Installing collected packages: PyYAML, pydub, pyaudio
Successfully installed PyYAML-6.0.1 pyaudio-0.2.14 pydub-0.25.1
Please provide your configuration values:
alsa_hw_mapping (default 1): 1
number of microphone channels (default 2): 2
recording sample rate (default 44100): 44100
recording format - INT16, INT32, FLOAT32 (default INT16): INT16
hook type - normally open (NO) or normally closed (default NC): NO
hook_gpio (default 22): 22
recording_limit - in seconds (default 300): 300
This will overwrite the current config.yaml. Are you sure? (Y/N): Y
Created symlink /etc/systemd/system/multi-user.target.wants/audioGuestBook.service → /etc/systemd/system/audioGuestBook.service.
Installation completed successfully!
admin@pi:~/rotary-phone-audio-guestbook $ CD
-bash: CD: command not found
admin@pi:~/rotary-phone-audio-guestbook $ cd
admin@pi:~ $ journalctl -u audioGuestBook -f
-- Journal begins at Tue 2023-12-05 11:07:43 +08. --
Feb 08 14:57:21 pi env[1337]:   File "/home/admin/rotary-phone-audio-guestbook/audioGuestBook.py", line 9, in <module>
Feb 08 14:57:21 pi env[1337]:     import pyaudio
Feb 08 14:57:21 pi env[1337]: ModuleNotFoundError: No module named 'pyaudio'
Feb 08 14:57:21 pi systemd[1]: audioGuestBook.service: Main process exited, code=exited, status=1/FAILURE
Feb 08 14:57:21 pi systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Feb 08 14:57:21 pi systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 5.
Feb 08 14:57:21 pi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 08 14:57:21 pi systemd[1]: audioGuestBook.service: Start request repeated too quickly.
Feb 08 14:57:21 pi systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Feb 08 14:57:21 pi systemd[1]: Failed to start Rotary Phone Guest Book Project`

error is `ModuleNotFoundError: No module named 'pyaudio'`. 

Please advise
Thank you

ipanggggg, 2024-02-08:
Hi @nickpourazima 

i just reinstall back pydub, pyaudio and pyyaml. Seems like module not found error has been gone. Now looks like it stuck at ```Started Rotary Phone Guest Book Project```

```
admin@pi:~ $ journalctl -u audioGuestBook -f
-- Journal begins at Tue 2023-12-05 11:07:43 +08. --
Feb 08 15:21:02 pi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 08 15:21:02 pi systemd[1]: audioGuestBook.service: Start request repeated too quickly.
Feb 08 15:21:02 pi systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Feb 08 15:21:02 pi systemd[1]: Failed to start Rotary Phone Guest Book Project.
-- Boot a8e73d1c8aa148fab9d3e504bd61dbb5 --
Feb 08 15:23:23 pi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 08 15:26:20 pi systemd[1]: Stopping Rotary Phone Guest Book Project...
Feb 08 15:26:20 pi systemd[1]: audioGuestBook.service: Succeeded.
Feb 08 15:26:20 pi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 08 15:26:20 pi systemd[1]: audioGuestBook.service: Consumed 3.614s CPU time.
-- Boot a383b702126d4e77bb72bf36023996dc --
Feb 08 15:26:30 pi systemd[1]: Started Rotary Phone Guest Book Project.

```

Please advise,
thank you

ipanggggg, 2024-02-11:
Hi @nickpourazima 

Im still no idea how to counter this issue. Im reinstall the raspbian and this repo more than 50 times with no success. Run ```sudo service audioGuestBook status``` and ```journalctl``` just show and stuck at ```Started Rotary Phone Guest Book Project``` only with no any error. Im also try to run script manually in directory but nothing happens. 

My setup is Raspberry Pi Zero 2W, Pi image 32bit lite.

Waiting your respond, im really appreciate it

nickpourazima, 2024-02-11:
Hi @ipanggggg, 

Would you be able to let me know your hardware connection set up? How did you configure the on/off hook gpio? Which rotary phone are you using? Were you able to prove it with a multimeter prior to check continuity?

It's unclear what the issue you're running into actually is, can you provide more details please?

ipanggggg, 2024-02-11:
Hi @nickpourazima 

Thank you for reply. Now above issue was solved. Im installing Pydub, pyaudio, pyyaml at the root instead of installation directory, thats why its not working hehe. Im not install it into rotary phone yet and im using momentary switch act as phone hook for temporary. 

Now its functioning half way. Its playing voicemail.wav when i pick up the hook but there are no beep sound 

```
-- Boot 86e0affb842649d496a42b9b206a8f4f --
Feb 12 02:52:10 pi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 12 02:53:53 pi env[575]: INFO:__main__:Phone off hook, ready to begin!
Feb 12 02:53:53 pi env[575]: INFO:__main__:Playing voicemail message...
Feb 12 02:53:53 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
Feb 12 02:53:53 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
Feb 12 02:53:53 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
Feb 12 02:53:53 pi env[575]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.USB-Audio.pcm.hdmi.0:CARD=1,AES0=4,AES1=130,AES2=0,AES3=2'
Feb 12 02:53:53 pi env[575]: ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Feb 12 02:53:53 pi env[575]: ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
Feb 12 02:53:53 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM hdmi
Feb 12 02:53:53 pi env[575]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.USB-Audio.pcm.hdmi.0:CARD=1,AES0=4,AES1=130,AES2=0,AES3=2'
Feb 12 02:53:53 pi env[575]: ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Feb 12 02:53:53 pi env[575]: ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
Feb 12 02:53:53 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM hdmi
Feb 12 02:53:53 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
Feb 12 02:53:53 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
Feb 12 02:53:53 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Feb 12 02:53:53 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Feb 12 02:53:53 pi env[575]: Cannot connect to server socket err = No such file or directory
Feb 12 02:53:53 pi env[575]: Cannot connect to server request channel
Feb 12 02:53:53 pi env[575]: jack server is not running or cannot be started
Feb 12 02:53:53 pi env[575]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Feb 12 02:53:53 pi env[575]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Feb 12 02:54:01 pi env[575]: INFO:__main__:Playing beep...
Feb 12 02:54:01 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
Feb 12 02:54:01 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
Feb 12 02:54:01 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
Feb 12 02:54:01 pi env[575]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.USB-Audio.pcm.hdmi.0:CARD=1,AES0=4,AES1=130,AES2=0,AES3=2'
Feb 12 02:54:01 pi env[575]: ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Feb 12 02:54:01 pi env[575]: ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
Feb 12 02:54:01 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM hdmi
Feb 12 02:54:01 pi env[575]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.USB-Audio.pcm.hdmi.0:CARD=1,AES0=4,AES1=130,AES2=0,AES3=2'
Feb 12 02:54:01 pi env[575]: ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Feb 12 02:54:01 pi env[575]: ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
Feb 12 02:54:01 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM hdmi
Feb 12 02:54:01 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
Feb 12 02:54:01 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
Feb 12 02:54:01 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Feb 12 02:54:01 pi env[575]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Feb 12 02:54:01 pi env[575]: Cannot connect to server socket err = No such file or directory
Feb 12 02:54:01 pi env[575]: Cannot connect to server request channel
Feb 12 02:54:01 pi env[575]: jack server is not running or cannot be started
Feb 12 02:54:01 pi env[575]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Feb 12 02:54:01 pi env[575]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
```

its show playing beep but no beep sound heared and its stuck there. When i put down and pick up back the hook, there are nothing happens. I can confirmed that beep.wav are exist in sound folder

nickpourazima, 2024-02-11:
What audio output are you trying to play through currently?
A few things to try:

- `pip install sounddevice`
- setup the default audio output device in `/etc/asound.conf` based on the numbers in `aplay -l`, it might look something like this:
```
pcm.!default {
    type hw
    card 2
    device 0
}

ctl.!default {
    type hw
    card 2
}
```

- set line 66 in audioInterface py to False: 
`data = self.stream.read(self.chunk, exception_on_overflow=False)`

ipanggggg, 2024-02-11:
I just try as per advise, seems like nothing changed. My audio output is using USB sound card already set as default card. Its Just a cheap soundcard and no datasheet available to see what sampling rate and bit depth it support.

[image]

below is log,  its playing voicemail.wav and i can hear it. Once its go to playing beep, i cannot hear any beep and it stop here, its not going to next step which is recording 

```
master@pi:~ $ journalctl -u audioGuestBook -f
-- Journal begins at Tue 2023-12-05 10:57:17 +08. --
Feb 12 07:29:28 pi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 12 07:30:08 pi env[579]: INFO:__main__:Phone off hook, ready to begin!
Feb 12 07:30:08 pi env[579]: INFO:__main__:Playing voicemail message...
Feb 12 07:30:17 pi env[579]: INFO:__main__:Playing beep...
```

i also notice that my voicemail.wav is played a bit speeded up. Looks like my cases is a bit similar with our friend that rising up the issue before named "Sound is garbled after first recording".

nickpourazima, 2024-02-12:
Hmm, okay. Can you do me a favor and checkout my dev branch and run the installer script?

```
cd <project root>
git checkout dev
git pull
chmod +x ./installer.sh
./installer.sh
```

Run through the installer and fill out the prompts. Let me know if this ends up working out for you, thanks!

ipanggggg, 2024-02-13:
Hi @nickpourazima , Sorry for delay.. Yesterday im just pulling your new updates but the issue still persists. However, today i reflashed back sd card and make a new clean install and wowww everything was working flawlessly now 😃. For this installation i do ```pip install sounddevice``` under project directory. Before this i do it at the root. Thank you for your great and effort to assist, will test it again

nickpourazima, 2024-02-18:
Yeah, I believe I'm not even utilizing the Jack server for anything anyways. Alrighty, glad it helped! I'll close this out.

closed 2024-02-18 by nickpourazima