nickpourazima/rotary-phone-audio-guestbook #23: something is wrong

s3b3n, 2024-02-18:
Hey,

hope you can help me. 
I installed everything and checked the soundcard. I can play sounds e.g. beep.wav and listen to it

I run  journalctl -u audioGuestBook -f
and get the following log:

-- Journal begins at Tue 2023-12-05 03:07:44 GMT. --
Feb 18 12:44:17 raspberrypi env[1298]:   File "/home/admin/rotary-phone-audio-guestbook/audioGuestBook.py", line 12, in <module>
Feb 18 12:44:17 raspberrypi env[1298]:     from pydub import AudioSegment, playback
Feb 18 12:44:17 raspberrypi env[1298]: ModuleNotFoundError: No module named 'pydub'
Feb 18 12:44:17 raspberrypi systemd[1]: audioGuestBook.service: Main process exited, code=exited, status=1/FAILURE
Feb 18 12:44:17 raspberrypi systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Feb 18 12:44:17 raspberrypi systemd[1]: audioGuestBook.service: Consumed 2.188s CPU time.
Feb 18 12:44:17 raspberrypi systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 163.
Feb 18 12:44:17 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 18 12:44:17 raspberrypi systemd[1]: audioGuestBook.service: Consumed 2.188s CPU time.

It seems that pydub is missing, I checked several times and pydub is installed.

Do you have any idea?

Kind regards
Sebastian

nickpourazima, 2024-02-18:
Hey @s3b3n, a few users were experiencing the same as you, try to checkout the dev branch and run the installer script and let me know if that resolves the issues. Another user has to run it on a clean install but we can leave that as a last resort. 

```
cd <project root>
git checkout dev
git pull
chmod +x ./installer.sh
./installer.sh
```

s3b3n, 2024-02-18:
Hey Nick,

thanks for the quick answer, i already tried the dev branch with a clean install on another sd card.
Problem exist nonetheless.

Do you have any idea what i can do ?

nickpourazima, 2024-02-18:
Hmmm, have you tried installing pydub locally with: `pip install --user pydub`?

s3b3n, 2024-02-18:
Now i have a new error:

-- Journal begins at Tue 2023-12-05 03:57:18 CET. --
Feb 18 17:14:02 raspberrypi env[2246]:   File "/home/admin/rotary-phone-audio-guestbook/src/audioGuestBook.py", line 14, in <module>
Feb 18 17:14:02 raspberrypi env[2246]:     import src.audioInterface as audioInterface
Feb 18 17:14:02 raspberrypi env[2246]: ModuleNotFoundError: No module named 'src'
Feb 18 17:14:02 raspberrypi systemd[1]: audioGuestBook.service: Main process exited, code=exited, status=1/FAILURE
Feb 18 17:14:02 raspberrypi systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Feb 18 17:14:02 raspberrypi systemd[1]: audioGuestBook.service: Consumed 2.136s CPU time.
Feb 18 17:14:02 raspberrypi systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 593.
Feb 18 17:14:02 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 18 17:14:02 raspberrypi systemd[1]: audioGuestBook.service: Consumed 2.136s CPU time.

Feb 18 17:14:02 raspberrypi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 18 17:14:04 raspberrypi env[2247]: /root/.local/lib/python3.9/site-packages/pydub/utils.py:170: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
Feb 18 17:14:04 raspberrypi env[2247]:   warn("Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work", RuntimeWarning)
Feb 18 17:14:04 raspberrypi env[2247]: Traceback (most recent call last):
Feb 18 17:14:04 raspberrypi env[2247]:   File "/home/admin/rotary-phone-audio-guestbook/src/audioGuestBook.py", line 14, in <module>
Feb 18 17:14:04 raspberrypi env[2247]:     import src.audioInterface as audioInterface
Feb 18 17:14:04 raspberrypi env[2247]: ModuleNotFoundError: No module named 'src'

nickpourazima, 2024-02-18:
Oh that's a bug in the dev branch, thanks for pointing it out. Change:
Line 14 in `audioGuestBook.py` from:
```
import src.audioInterface as audioInterface
```
To:
```
import audioInterface as audioInterface
```

nickpourazima, 2024-02-18:
Also, double check that you've got ffmpeg installed, if not you can do a:

```
sudo apt update
sudo apt install ffmpeg
```

s3b3n, 2024-02-18:
Did everything you mentioned now there is a new error:
`admin@raspberrypi:~/rotary-phone-audio-guestbook/src $ journalctl -u audioGuestBook -f
-- Journal begins at Tue 2023-12-05 03:57:18 CET. --
Feb 18 17:45:16 raspberrypi env[4124]:   File "/root/.local/lib/python3.9/site-packages/pydub/scipy_effects.py", line 12, in <module>
Feb 18 17:45:16 raspberrypi env[4124]:     from scipy.signal import butter, sosfilt
Feb 18 17:45:16 raspberrypi env[4124]: ModuleNotFoundError: No module named 'scipy'
Feb 18 17:45:16 raspberrypi systemd[1]: audioGuestBook.service: Main process exited, code=exited, status=1/FAILURE
Feb 18 17:45:16 raspberrypi systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Feb 18 17:45:16 raspberrypi systemd[1]: audioGuestBook.service: Consumed 2.136s CPU time.
Feb 18 17:45:16 raspberrypi systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 1266.
Feb 18 17:45:16 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 18 17:45:16 raspberrypi systemd[1]: audioGuestBook.service: Consumed 2.136s CPU time.
Feb 18 17:45:16 raspberrypi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 18 17:45:18 raspberrypi env[4125]: Traceback (most recent call last):
Feb 18 17:45:18 raspberrypi env[4125]:   File "/home/admin/rotary-phone-audio-guestbook/src/audioGuestBook.py", line 14, in <module>
Feb 18 17:45:18 raspberrypi env[4125]:     import audioInterface as audioInterface
Feb 18 17:45:18 raspberrypi env[4125]:   File "/home/admin/rotary-phone-audio-guestbook/src/audioInterface.py", line 10, in <module>
Feb 18 17:45:18 raspberrypi env[4125]:     from pydub.scipy_effects import band_pass_filter
`

nickpourazima, 2024-02-18:
I just pushed some changes, check it out w/ a `git pull`. Scipy is only used in the audioProcessing functions. Since that stuff is optional and unused currently in the main code I moved it to its own class. I also updated the installer to be inside a standalone virtual environment so we won't need to worry about --user vs. admin package issues.

If it doesn't directly work with these updates try running through the new installer.

s3b3n, 2024-02-18:
I removed the current Porject folder and pulled the newest code from the dev branch.
Two errors occurred:
First:
The installation script works fine but throws this error at the end.
admin@raspberrypi:~/rotary-phone-audio-guestbook $ ./installer.sh
Starting the installation process...
Installing additional dependencies...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-venv is already the newest version (3.9.2-3).
ffmpeg is already the newest version (8:4.3.6-0+deb11u1+rpt5).
python3-gpiozero is already the newest version (1.6.2-1).
python3-pip is already the newest version (20.3.4-4+rpt1+deb11u1).
0 upgraded, 0 newly installed, 0 to remove and 23 not upgraded.
Setting up Python virtual environment...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pydub in /home/admin/rotary-phone-venv/lib/python3.9/site-packages (0.25.1)
Requirement already satisfied: pyaudio in /home/admin/rotary-phone-venv/lib/python3.9/site-packages (0.2.14)
Requirement already satisfied: PyYAML in /home/admin/rotary-phone-venv/lib/python3.9/site-packages (6.0.1)
Requirement already satisfied: sounddevice in /home/admin/rotary-phone-venv/lib/python3.9/site-packages (0.4.6)
Requirement already satisfied: CFFI>=1.0 in /home/admin/rotary-phone-venv/lib/python3.9/site-packages (from sounddevice) (1.16.0)
Requirement already satisfied: pycparser in /home/admin/rotary-phone-venv/lib/python3.9/site-packages (from CFFI>=1.0->sounddevice) (2.21)
Configuring PulseAudio...
default-fragments = 5
default-fragment-size-msec = 2
./installer.sh: line 34: pulseaudio: command not found
./installer.sh: line 35: pulseaudio: command not found
Listing available sound cards and devices:
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
Configuring ALSA...
Enter the card number for the default playback card (e.g., 0, 1): 1
Enter the card number for the default capture card (e.g., 0, 1): 1
Enter the default sample rate (e.g., 44100): 44100
Enter the bit depth (16, 24, 32): 16
Applying ALSA configuration...
Testing recording and playback...
Recording WAVE 'test-mic.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
arecord: set_params:1349: Channels count non available
**Test failed. Check your microphone and speaker setup.**

Second:
admin@raspberrypi:~/rotary-phone-audio-guestbook $ journalctl -u audioGuestBook -f
-- Journal begins at Tue 2023-12-05 03:57:18 CET. --
Feb 18 18:03:15 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 18 18:03:15 raspberrypi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 18 18:03:15 raspberrypi env[4589]: python3: can't open file '/home/admin/rotary-phone-audio-guestbook/src/audioGuestBook.py': [Errno 2] No such file or directory
Feb 18 18:03:15 raspberrypi systemd[1]: audioGuestBook.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Feb 18 18:03:15 raspberrypi systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Feb 18 18:03:15 raspberrypi systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 1700.
Feb 18 18:03:15 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 18 18:03:15 raspberrypi systemd[1]: audioGuestBook.service: Start request repeated too quickly.
Feb 18 18:03:15 raspberrypi systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Feb 18 18:03:15 raspberrypi systemd[1]: Failed to start Rotary Phone Guest Book Project.

nickpourazima, 2024-02-18:
Thank you for testing this and for your patience! I haven't had my rotary phone set back up so I'm flying a bit blind here.

1. Let's ensure pulseaudio is installed:
```
sudo apt-get update
sudo apt-get install -y pulseaudio
```

2. Don't worry about the sound test, I think I need to configure the channel differently.

3. Can you let me know what the output of the `audioGuestBook.service` is?
`cat /etc/systemd/system/audioGuestBook.service`

s3b3n, 2024-02-18:
Hey Nick,

i have to thank you for the great work and help.
I think your code will help others to build their own audio guestbook and save some money for their wedding.

Heres the output:

admin@raspberrypi:~/rotary-phone-audio-guestbook/sounds $ cat /etc/systemd/system/audioGuestBook.service
[Unit]
Description=Rotary Phone Guest Book Project
After=multi-user.target

[Service]
WorkingDirectory=/home/admin/rotary-phone-audio-guestbook
Type=simple
Restart=always
ExecStart=/usr/bin/env python3 /home/admin/rotary-phone-audio-guestbook/src/audioGuestBook.py

[Install]
WantedBy=multi-user.targetadmin@raspberrypi:~/rotary-phone-audio-guestbook/sounds $

nickpourazima, 2024-02-18:
No problem, glad I could help!

Is that a type on your `WantedBy` at the end there? It should just be:
`WantedBy=multi-user.target`

Also, let's try a couple things:

1. Let's make sure your audioGuestBook.py has executable permissions:
```
chmod +x /home/admin/rotary-phone-audio-guestbook/src/audioGuestBook.py
```

2. Since we changed to a virtual environment it might not be executing the service properly, try:
```
sudo sed -i "s|ExecStart=.*|ExecStart=/bin/bash -c 'source /home/admin/rotary-phone-venv/bin/activate && python3 /home/admin/rotary-phone-audio-guestbook/src/audioGuestBook.py'|" /etc/systemd/system/audioGuestBook.service
sudo systemctl daemon-reload
sudo systemctl restart audioGuestBook.service
```

s3b3n, 2024-02-18:
Yes this was added from copy and paste, it ende with 
`WantedBy=multi-user.target`

Now we have a strange error:
`-- Journal begins at Tue 2023-12-05 03:57:18 CET. --
Feb 18 18:49:52 raspberrypi systemd[1]: audioGuestBook.service: Succeeded.
Feb 18 18:49:53 raspberrypi systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 4.
Feb 18 18:49:53 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 18 18:49:53 raspberrypi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 18 18:49:53 raspberrypi systemd[1]: audioGuestBook.service: Succeeded.
Feb 18 18:49:53 raspberrypi systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 5.
Feb 18 18:49:53 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 18 18:49:53 raspberrypi systemd[1]: audioGuestBook.service: Start request repeated too quickly.
Feb 18 18:49:53 raspberrypi systemd[1]: audioGuestBook.service: Failed with result 'start-limit-hit'.
Feb 18 18:49:53 raspberrypi systemd[1]: Failed to start Rotary Phone Guest Book Project.`

nickpourazima, 2024-02-18:
Was that the output from: `journalctl -u audioGuestBook.service -e`?

s3b3n, 2024-02-18:
journalctl -u audioGuestBook.service -e
Feb 18 19:27:08 raspberrypi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 18 19:27:08 raspberrypi systemd[1]: audioGuestBook.service: Succeeded.
Feb 18 19:27:08 raspberrypi systemd[1]: audioGuestBook.service: Scheduled resta>
Feb 18 19:27:08 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 18 19:27:08 raspberrypi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 18 19:27:08 raspberrypi systemd[1]: audioGuestBook.service: Succeeded.
Feb 18 19:27:09 raspberrypi systemd[1]: audioGuestBook.service: Scheduled resta>
Feb 18 19:27:09 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 18 19:27:09 raspberrypi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 18 19:27:09 raspberrypi systemd[1]: audioGuestBook.service: Succeeded.
Feb 18 19:27:09 raspberrypi systemd[1]: audioGuestBook.service: Scheduled resta>
Feb 18 19:27:09 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 18 19:27:09 raspberrypi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 18 19:27:09 raspberrypi systemd[1]: audioGuestBook.service: Succeeded.
Feb 18 19:27:09 raspberrypi systemd[1]: audioGuestBook.service: Scheduled resta>
Feb 18 19:27:09 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 18 19:27:09 raspberrypi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 18 19:27:09 raspberrypi systemd[1]: audioGuestBook.service: Succeeded.
Feb 18 19:27:09 raspberrypi systemd[1]: audioGuestBook.service: Scheduled resta>
Feb 18 19:27:09 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 18 19:27:09 raspberrypi systemd[1]: audioGuestBook.service: Start request r>
Feb 18 19:27:09 raspberrypi systemd[1]: audioGuestBook.service: Failed with res>
Feb 18 19:27:09 raspberrypi systemd[1]: Failed to start Rotary Phone Guest Book>

nickpourazima, 2024-02-18:
Try updating your audioGuestBook.service file to this using `sudo nano /etc/systemd/system/audioGuestBook.service` or whatever flavor of text editor:

```
[Service]
WorkingDirectory=/home/admin/rotary-phone-audio-guestbook
Type=simple
Restart=always
ExecStart=/home/admin/rotary-phone-venv/bin/python3 /home/admin/rotary-phone-audio-guestbook/src/audioGuestBook.py
```

Save it and then reload/restart the service:
```
sudo systemctl daemon-reload
sudo systemctl restart audioGuestBook.service
```

s3b3n, 2024-02-19:
Hey Nick,

sry for the late answer, i tried your suggestions.
Now i get this error:
admin@raspberrypi:~ $ journalctl -u audioGuestBook.service -e
Feb 19 20:27:45 raspberrypi systemd[1]: audioGuestBook.service: Main process exited, code=exited, status=1/FAILURE
Feb 19 20:27:45 raspberrypi systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Feb 19 20:27:45 raspberrypi systemd[1]: audioGuestBook.service: Consumed 1.252s CPU time.
Feb 19 20:27:45 raspberrypi systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 3.
Feb 19 20:27:45 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 19 20:27:45 raspberrypi systemd[1]: audioGuestBook.service: Consumed 1.252s CPU time.
Feb 19 20:27:45 raspberrypi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 19 20:27:47 raspberrypi python3[919]: Traceback (most recent call last):
Feb 19 20:27:47 raspberrypi python3[919]:   File "/home/admin/rotary-phone-audio-guestbook/src/audioGuestBook.py", li>
Feb 19 20:27:47 raspberrypi python3[919]:     from gpiozero import Button
Feb 19 20:27:47 raspberrypi python3[919]: ModuleNotFoundError: No module named 'gpiozero'
Feb 19 20:27:47 raspberrypi systemd[1]: audioGuestBook.service: Main process exited, code=exited, status=1/FAILURE
Feb 19 20:27:47 raspberrypi systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Feb 19 20:27:47 raspberrypi systemd[1]: audioGuestBook.service: Consumed 1.308s CPU time.
Feb 19 20:27:47 raspberrypi systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 4.
Feb 19 20:27:47 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 19 20:27:47 raspberrypi systemd[1]: audioGuestBook.service: Consumed 1.308s CPU time.
Feb 19 20:27:47 raspberrypi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 19 20:27:48 raspberrypi python3[920]: Traceback (most recent call last):
Feb 19 20:27:48 raspberrypi python3[920]:   File "/home/admin/rotary-phone-audio-guestbook/src/audioGuestBook.py", li>
Feb 19 20:27:48 raspberrypi python3[920]:     from gpiozero import Button
Feb 19 20:27:48 raspberrypi python3[920]: ModuleNotFoundError: No module named 'gpiozero'
Feb 19 20:27:48 raspberrypi systemd[1]: audioGuestBook.service: Main process exited, code=exited, status=1/FAILURE
Feb 19 20:27:48 raspberrypi systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Feb 19 20:27:48 raspberrypi systemd[1]: audioGuestBook.service: Consumed 1.326s CPU time.
Feb 19 20:27:49 raspberrypi systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 5.
Feb 19 20:27:49 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 19 20:27:49 raspberrypi systemd[1]: audioGuestBook.service: Consumed 1.326s CPU time.
Feb 19 20:27:49 raspberrypi systemd[1]: audioGuestBook.service: Start request repeated too quickly.
Feb 19 20:27:49 raspberrypi systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Feb 19 20:27:49 raspberrypi systemd[1]: Failed to start Rotary Phone Guest Book Project.

Do you have any idea?

nickpourazima, 2024-02-23:
Hmm, that's odd. So we know that python3-gpiozero is installed globally but since we're now using a virtual env it might be necessary to install within there as well, try:
```bash
source ~/rotary-phone-venv/bin/activate
pip install gpiozero
```

s3b3n, 2024-02-23:
The installation worked but i get the same error:

 journalctl -u audioGuestBook.service -                                      e
Feb 23 07:36:17 raspberrypi python3[927]:   File "/home/admin/rotary-phone-audi>
Feb 23 07:36:17 raspberrypi python3[927]:     from gpiozero import Button
Feb 23 07:36:17 raspberrypi python3[927]: ModuleNotFoundError: No module named >
Feb 23 07:36:17 raspberrypi systemd[1]: audioGuestBook.service: Main process ex>
Feb 23 07:36:17 raspberrypi systemd[1]: audioGuestBook.service: Failed with res>
Feb 23 07:36:17 raspberrypi systemd[1]: audioGuestBook.service: Consumed 1.262s>
Feb 23 07:36:17 raspberrypi systemd[1]: audioGuestBook.service: Scheduled resta>
Feb 23 07:36:17 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 23 07:36:17 raspberrypi systemd[1]: audioGuestBook.service: Consumed 1.262s>
Feb 23 07:36:17 raspberrypi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 23 07:36:19 raspberrypi python3[930]: Traceback (most recent call last):
Feb 23 07:36:19 raspberrypi python3[930]:   File "/home/admin/rotary-phone-audi>
Feb 23 07:36:19 raspberrypi python3[930]:     from gpiozero import Button
Feb 23 07:36:19 raspberrypi python3[930]: ModuleNotFoundError: No module named >
Feb 23 07:36:19 raspberrypi systemd[1]: audioGuestBook.service: Main process ex>
Feb 23 07:36:19 raspberrypi systemd[1]: audioGuestBook.service: Failed with res>
Feb 23 07:36:19 raspberrypi systemd[1]: audioGuestBook.service: Consumed 1.266s>
Feb 23 07:36:19 raspberrypi systemd[1]: audioGuestBook.service: Scheduled resta>
Feb 23 07:36:19 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 23 07:36:19 raspberrypi systemd[1]: audioGuestBook.service: Consumed 1.266s>
Feb 23 07:36:19 raspberrypi systemd[1]: audioGuestBook.service: Start request r>
Feb 23 07:36:19 raspberrypi systemd[1]: audioGuestBook.service: Failed with res>
Feb 23 07:36:19 raspberrypi systemd[1]: Failed to start Rotary Phone Guest Book>

s3b3n, 2024-02-24:
Hey Nick,
today i tried again and booted up the raspberry pi, now the gpiozero error is gone but i got a new error that the config.yaml file was missing in the /src folder. I moved it in this location and now i get this new error messages:

admin@raspberrypi:~/rotary-phone-audio-guestbook/sounds $ journalctl -u audioGuestBook.service -e
Feb 24 12:55:30 raspberrypi python3[1143]: /home/admin/rotary-phone-venv/lib/python3.9/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from pigpio: No module named >
Feb 24 12:55:30 raspberrypi python3[1143]:   warnings.warn(
Feb 24 12:55:30 raspberrypi python3[1143]: /home/admin/rotary-phone-venv/lib/python3.9/site-packages/gpiozero/devices.py:297: NativePinFactoryFallback: Falling back to the experimental pin>
Feb 24 12:55:30 raspberrypi python3[1143]:   warnings.warn(NativePinFactoryFallback(native_fallback_message))
Feb 24 12:56:10 raspberrypi systemd[1]: Stopping Rotary Phone Guest Book Project...
Feb 24 12:56:10 raspberrypi systemd[1]: audioGuestBook.service: Succeeded.
Feb 24 12:56:10 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 24 12:56:10 raspberrypi systemd[1]: audioGuestBook.service: Consumed 4.629s CPU time.
Feb 24 12:56:11 raspberrypi systemd[1]: Started Rotary Phone Guest Book Project.
Feb 24 12:56:13 raspberrypi python3[1180]: INFO:__main__:Remember to monitor system resources during recording.
Feb 24 12:56:13 raspberrypi python3[1180]: /home/admin/rotary-phone-venv/lib/python3.9/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from lgpio: No module named '>
Feb 24 12:56:13 raspberrypi python3[1180]:   warnings.warn(
Feb 24 12:56:13 raspberrypi python3[1180]: /home/admin/rotary-phone-venv/lib/python3.9/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from rpigpio: No module named>
Feb 24 12:56:13 raspberrypi python3[1180]:   warnings.warn(
Feb 24 12:56:13 raspberrypi python3[1180]: /home/admin/rotary-phone-venv/lib/python3.9/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from pigpio: No module named >
Feb 24 12:56:13 raspberrypi python3[1180]:   warnings.warn(
Feb 24 12:56:13 raspberrypi python3[1180]: /home/admin/rotary-phone-venv/lib/python3.9/site-packages/gpiozero/devices.py:297: NativePinFactoryFallback: Falling back to the experimental pin>
Feb 24 12:56:13 raspberrypi python3[1180]:   warnings.warn(NativePinFactoryFallback(native_fallback_message))
Feb 24 12:59:04 raspberrypi systemd[1]: Stopping Rotary Phone Guest Book Project...
Feb 24 12:59:04 raspberrypi systemd[1]: audioGuestBook.service: Succeeded.
Feb 24 12:59:04 raspberrypi systemd[1]: Stopped Rotary Phone Guest Book Project.
Feb 24 12:59:04 raspberrypi systemd[1]: audioGuestBook.service: Consumed 10.381s CPU time.

I think we are on a good way do you have any ideas, what i can try?

s3b3n, 2024-02-24:
Adter reboot i get  a new error log:

admin@raspberrypi:~/rotary-phone-audio-guestbook $ journalctl -u audioGuestBook.service -e
Feb 24 19:13:10 raspberrypi python3[1210]: jack server is not running or cannot be started
Feb 24 19:13:10 raspberrypi python3[1210]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Feb 24 19:13:10 raspberrypi python3[1210]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Feb 24 19:13:10 raspberrypi python3[1210]: ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
Feb 24 19:13:10 raspberrypi python3[1210]: ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
Feb 24 19:13:10 raspberrypi python3[1210]: ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection refused
Feb 24 19:13:10 raspberrypi python3[1210]: ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection refused
Feb 24 19:13:11 raspberrypi python3[1210]: ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
Feb 24 19:13:11 raspberrypi python3[1210]: ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
Feb 24 19:13:11 raspberrypi python3[1210]: ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
Feb 24 19:13:11 raspberrypi python3[1210]: Cannot connect to server socket err = No such file or directory
Feb 24 19:13:11 raspberrypi python3[1210]: Cannot connect to server request channel
Feb 24 19:13:11 raspberrypi python3[1210]: jack server is not running or cannot be started
Feb 24 19:13:11 raspberrypi python3[1210]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Feb 24 19:13:11 raspberrypi python3[1210]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Feb 24 19:13:11 raspberrypi python3[1210]: INFO:audioInterface:Audio resources initialized.
Feb 24 19:13:11 raspberrypi python3[1210]: INFO:audioInterface:Audio stream initialized for recording.
Feb 24 19:13:11 raspberrypi python3[1210]: Expression 'parameters->channelCount <= maxChans' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1514
Feb 24 19:13:11 raspberrypi python3[1210]: Expression 'ValidateParameters( inputParameters, hostApi, StreamDirection_In )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2818
Feb 24 19:13:11 raspberrypi python3[1210]: Exception in thread Thread-1:
Feb 24 19:13:11 raspberrypi python3[1210]: Traceback (most recent call last):
Feb 24 19:13:11 raspberrypi python3[1210]:   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
Feb 24 19:13:11 raspberrypi python3[1210]:     self.run()
Feb 24 19:13:11 raspberrypi python3[1210]:   File "/home/admin/rotary-phone-venv/lib/python3.9/site-packages/lgpio.py", line 557, in run
Feb 24 19:13:11 raspberrypi python3[1210]:     cb.func(chip, gpio, level, tick)
Feb 24 19:13:11 raspberrypi python3[1210]:   File "/home/admin/rotary-phone-venv/lib/python3.9/site-packages/gpiozero/pins/lgpio.py", line 248, in _call_when_changed
Feb 24 19:13:11 raspberrypi python3[1210]:     super()._call_when_changed(ticks / 1000000000, level)
Feb 24 19:13:11 raspberrypi python3[1210]:   File "/home/admin/rotary-phone-venv/lib/python3.9/site-packages/gpiozero/pins/local.py", line 111, in _call_when_changed
Feb 24 19:13:11 raspberrypi python3[1210]:     super()._call_when_changed(
Feb 24 19:13:11 raspberrypi python3[1210]:   File "/home/admin/rotary-phone-venv/lib/python3.9/site-packages/gpiozero/pins/pi.py", line 618, in _call_when_changed
Feb 24 19:13:11 raspberrypi python3[1210]:     method(ticks, state)
Feb 24 19:13:11 raspberrypi python3[1210]:   File "/home/admin/rotary-phone-venv/lib/python3.9/site-packages/gpiozero/input_devices.py", line 179, in _pin_changed
Feb 24 19:13:11 raspberrypi python3[1210]:     self._fire_events(ticks, bool(self._state_to_value(state)))
Feb 24 19:13:11 raspberrypi python3[1210]:   File "/home/admin/rotary-phone-venv/lib/python3.9/site-packages/gpiozero/mixins.py", line 389, in _fire_events
Feb 24 19:13:11 raspberrypi python3[1210]:     self._fire_deactivated()
Feb 24 19:13:11 raspberrypi python3[1210]:   File "/home/admin/rotary-phone-venv/lib/python3.9/site-packages/gpiozero/mixins.py", line 436, in _fire_deactivated
Feb 24 19:13:11 raspberrypi python3[1210]:     super()._fire_deactivated()
Feb 24 19:13:11 raspberrypi python3[1210]:   File "/home/admin/rotary-phone-venv/lib/python3.9/site-packages/gpiozero/mixins.py", line 353, in _fire_deactivated
Feb 24 19:13:11 raspberrypi python3[1210]:     self.when_deactivated()
Feb 24 19:13:11 raspberrypi python3[1210]:   File "/home/admin/rotary-phone-audio-guestbook/src/audioGuestBook.py", line 98, in off_hook
Feb 24 19:13:11 raspberrypi python3[1210]:     audio_interface.record()
Feb 24 19:13:11 raspberrypi python3[1210]:   File "/home/admin/rotary-phone-audio-guestbook/src/audioInterface.py", line 123, in record
Feb 24 19:13:11 raspberrypi python3[1210]:     self.stream = self.audio.open(
Feb 24 19:13:11 raspberrypi python3[1210]:   File "/home/admin/rotary-phone-venv/lib/python3.9/site-packages/pyaudio/__init__.py", line 639, in open
Feb 24 19:13:11 raspberrypi python3[1210]:     stream = PyAudio.Stream(self, *args, **kwargs)
Feb 24 19:13:11 raspberrypi python3[1210]:   File "/home/admin/rotary-phone-venv/lib/python3.9/site-packages/pyaudio/__init__.py", line 441, in __init__
Feb 24 19:13:11 raspberrypi python3[1210]:     self._stream = pa.open(**arguments)
Feb 24 19:13:11 raspberrypi python3[1210]: OSError: [Errno -9998] Invalid number of channels

nickpourazima, 2024-02-29:
@s3b3n I think I know what was wrong, can you pull the latest dev and re-test?

The issue appears to be due to using the `wave.Wave_write` object in a `with` statement without it being a context manager by default. 

The wave module's `Wave_write` class doesn't support context management out of the box, which means it lacks the ` __enter__()` and `__exit__()` methods necessary for the with statement to function as intended for automatic resource management.

To fix it I am manually managing the opening and closing of the `Wave_write` object without relying on the with statement for context management. 

Reference: https://stackoverflow.com/questions/46538867/attributeerror-wave-write-instance-has-no-attribute-exit

s3b3n, 2024-03-03:
Hey @nickpourazima,

sry for the late response. Today i did a fresh install with the newest dev branch.
Sadly there were some issues that we have to fix in the installer script:

1. During the installaton there is an error in pulseaudio
` E: [pulseaudio] main.c: Failed to kill daemon: No such process`2
2. Sound files are located in `/sounds` the service is searching in `/src/sounds`
3.  gpiozero is missing in venv, to fix this we need to add following install lines to the script:
```
source ~/rotary-phone-venv/bin/activate
pip install gpiozero
```
4.` config.yaml` is located in `/config.yaml` the service is searching in `/src/config.yaml`
5.The lg libary is missing (http://abyz.me.uk/lg/download.html) as mentioned abov three files from this libary has to be copied in the venv because of this issue (https://github.com/gpiozero/gpiozero/issues/1038)
We can install the lg libary with the following commands 
```
wget http://abyz.me.uk/lg/lg.zip
unzip lg.zip
cd lg
make
sudo make install
```
The files are we need are 
`lgpio.py` located in `/lg/PY_LGPIO`
`lgpio.egg-info/` located in  `/lg/PY_LGPIO`
`_lgpio.cpython-39-arm-linux-gnueabihf.so` located in `/lg/PY_LGPIO/build/lib.linux-armv6l-3.9`

These files have to be copied to this directory `~/rotary-phone-venv/lib/python3.9/site-packages/ `

After i manually adjusted the point above i get the same error as mentioned above and i cant record any messages:
```
Mar 03 13:34:47 raspberrypi python3[833]: INFO:__main__:Phone off hook, ready to begin!
Mar 03 13:34:47 raspberrypi python3[833]: INFO:audioInterface:Initializing Audio Interface with sample rate: 44100, format: 8
Mar 03 13:34:47 raspberrypi python3[833]: INFO:audioInterface:Audio resources initialized.
Mar 03 13:34:47 raspberrypi python3[833]: INFO:__main__:Playing voicemail message...
Mar 03 13:34:53 raspberrypi python3[833]: INFO:__main__:Playing beep...
Mar 03 13:34:53 raspberrypi python3[833]: Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2048
Mar 03 13:34:53 raspberrypi python3[833]: Expression 'PaAlsaStreamComponent_InitialConfigure( &self->playback, outParams, self->primeBuffers, hwParamsPlayback, &realSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2721
Mar 03 13:34:53 raspberrypi python3[833]: Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2842
Mar 03 13:34:53 raspberrypi python3[833]: ERROR:__main__:Error playing beep.wav. Error: [Errno -9997] Invalid sample rate
Mar 03 13:34:53 raspberrypi python3[833]: INFO:__main__:Recording
Mar 03 13:34:53 raspberrypi python3[833]: INFO:audioInterface:Audio resources initialized.
Mar 03 13:34:53 raspberrypi python3[833]: INFO:audioInterface:Audio stream initialized for recording.
Mar 03 13:34:53 raspberrypi python3[833]: INFO:audioInterface:Stopping audio stream.
Mar 03 13:34:53 raspberrypi python3[833]: INFO:audioInterface:Terminating PyAudio session.
Mar 03 13:34:53 raspberrypi python3[833]: ERROR:audioInterface:Error writing to file /home/admin/rotary-phone-audio-guestbook/src/recordings/2024-03-03T13:34:53.957802.wav. Error: [Errno 2] No such file or directory: '/home/admin/rotary-phone-audio-guestbook/src/recordings/2024-03-03T13:34:53.957802.wav'
Mar 03 13:34:53 raspberrypi python3[833]: Exception ignored in: <function Wave_write.__del__ at 0xb5dcfd18>
Mar 03 13:34:53 raspberrypi python3[833]: Traceback (most recent call last):
Mar 03 13:34:53 raspberrypi python3[833]:   File "/usr/lib/python3.9/wave.py", line 326, in __del__
Mar 03 13:34:53 raspberrypi python3[833]:     self.close()
Mar 03 13:34:53 raspberrypi python3[833]:   File "/usr/lib/python3.9/wave.py", line 443, in close
Mar 03 13:34:54 raspberrypi python3[833]:     if self._file:
Mar 03 13:34:54 raspberrypi python3[833]: AttributeError: 'Wave_write' object has no attribute '_file'
Mar 03 13:34:54 raspberrypi python3[833]: INFO:__main__:Finished recording!
Mar 03 13:34:54 raspberrypi python3[833]: INFO:__main__:Phone on hook.
```

So your fix doesnt seem to work for me.

closed 2024-03-10 by nickpourazima