nickpourazima/rotary-phone-audio-guestbook #24: Can't get it to work

Andys280, 2024-03-04:
Hello . On a raspberry pi zero with freshly installed raspbian I clone the code, everything works ok in the installation, without errors. I answer the questions asked with the default answers and I receive the message that the installation completed successfully.
I connected a switch to the gpio pin, but when I press it, nothing happens. I tried to run the command `aplay -l` . My audio card on the usb port is the second one listed, so I changed the aplay configuration, which for me can be found at the address `/user/share/alsa/alsa.conf` (at the addresses indicated in the guide, I do not find any files). 

```
admin@Guestbook:~/rotary-phone-audio-guestbook $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
I run the command `sudo nano /usr/share/alsa/alsa.conf` and change the number from 0 to 1 on the following lines ```
defaults.pcm.card X
defaults.ctl.card X
```
Now i run the command `sudo /etc/init.d/alsa-utils restart` but again, nothing works.

This is what the log command shows me
```
admin@Guestbook:~/rotary-phone-audio-guestbook $ journalctl -u audioGuestBook -f
-- Journal begins at Tue 2023-12-05 03:07:44 GMT. --
Mar 04 12:19:01 Guestbook env[1704]:   File "/root/.local/lib/python3.9/site-packages/pyaudio/__init__.py", line 111, in <module>
Mar 04 12:19:01 Guestbook env[1704]:     import pyaudio._portaudio as pa
Mar 04 12:19:01 Guestbook env[1704]: ImportError: libportaudio.so.2: cannot open shared object file: No such file or directory
Mar 04 12:19:01 Guestbook systemd[1]: audioGuestBook.service: Main process exited, code=exited, status=1/FAILURE
Mar 04 12:19:01 Guestbook systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Mar 04 12:19:01 Guestbook systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 5.
Mar 04 12:19:01 Guestbook systemd[1]: Stopped Rotary Phone Guest Book Project.
Mar 04 12:19:01 Guestbook systemd[1]: audioGuestBook.service: Start request repeated too quickly.
Mar 04 12:19:01 Guestbook systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Mar 04 12:19:01 Guestbook systemd[1]: Failed to start Rotary Phone Guest Book Project.

```

Now I tried to install pydub locally and it installed succesfully with this command `pip install --user pydub`

Now i get a new error and i don t know how to repair it 
```
admin@Guestbook:~ $ journalctl -u audioGuestBook -f
-- Journal begins at Tue 2023-12-05 03:07:44 GMT. --
Mar 04 12:32:42 Guestbook env[837]:   File "/home/admin/rotary-phone-audio-guestbook/audioGuestBook.py", line 9, in <module>
Mar 04 12:32:42 Guestbook env[837]:     import pyaudio
Mar 04 12:32:42 Guestbook env[837]:   File "/root/.local/lib/python3.9/site-packages/pyaudio/__init__.py", line 111, in <module>
Mar 04 12:32:42 Guestbook env[837]:     import pyaudio._portaudio as pa
Mar 04 12:32:42 Guestbook env[837]: ImportError: libportaudio.so.2: cannot open shared object file: No such file or directory
Mar 04 12:32:43 Guestbook systemd[1]: audioGuestBook.service: Main process exited, code=exited, status=1/FAILURE
Mar 04 12:32:43 Guestbook systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Mar 04 12:32:43 Guestbook systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 18.
Mar 04 12:32:43 Guestbook systemd[1]: Stopped Rotary Phone Guest Book Project.
Mar 04 12:32:43 Guestbook systemd[1]: Started Rotary Phone Guest Book Project.
Mar 04 12:32:45 Guestbook env[842]: Could not import the PyAudio C module 'pyaudio._portaudio'.
Mar 04 12:32:45 Guestbook env[842]: Traceback (most recent call last):
Mar 04 12:32:45 Guestbook env[842]:   File "/home/admin/rotary-phone-audio-guestbook/audioGuestBook.py", line 9, in <module>
Mar 04 12:32:45 Guestbook env[842]:     import pyaudio
Mar 04 12:32:45 Guestbook env[842]:   File "/root/.local/lib/python3.9/site-packages/pyaudio/__init__.py", line 111, in <module>
Mar 04 12:32:45 Guestbook env[842]:     import pyaudio._portaudio as pa
Mar 04 12:32:45 Guestbook env[842]: ImportError: libportaudio.so.2: cannot open shared object file: No such file or directory
Mar 04 12:32:45 Guestbook systemd[1]: audioGuestBook.service: Main process exited, code=exited, status=1/FAILURE
Mar 04 12:32:45 Guestbook systemd[1]: audioGuestBook.service: Failed with result 'exit-code'.
Mar 04 12:32:45 Guestbook systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 19.
Mar 04 12:32:45 Guestbook systemd[1]: Stopped Rotary Phone Guest Book Project.

```

s3b3n, 2024-03-07:
Maybe try to install 
`pip install -- user pyaudio`

Andys280, 2024-03-07:
> Maybe try to install `pip install -- user pyaudio`

Can't run your command
```
admin@Guestbook:~ $ pip install -- user pyaudio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement user
ERROR: No matching distribution found for user
```

s3b3n, 2024-03-07:
Sry i had an error in the Syntax 
Try `pip install --user pyaudio`

Andys280, 2024-03-07:
> Sry i had an error in the Syntax Try `pip install --user pyaudio`

`Requirement already satisfied: pyaudio in /home/admin/.local/lib/python3.9/site-packages (0.2.14)`

Andys280, 2024-03-08:
I ve installed multiple things and now it starts without errrors , but not working. i can t hear anything in the speakers and no recording is made. i ve even placed an "voicemaile.wav" file in sounds folder. Tested sound card with aplay commands ad works . What can be the problem now ?

s3b3n, 2024-03-09:
What is the output from 
`journalctl -u audioGuestBook -f`

s3b3n, 2024-03-10:
@nickpourazima 

For me it works flawlessly, thank you for the great work.
One thing i personally would like to have is the option to configure the volume from every wav seperatly via config.yaml 

So everyone can fine tune how loud the beep or the voicemai is.

Great work, i will buy you a coffe for this ;)

nickpourazima, 2024-03-10:
Thanks @s3b3n! Glad it worked for you!

I can modify the scripts to accommodate volume levels per wav file. I'll probably call `amixer` prior to `aplay` with whatever is specified in the `config.yaml`. I'll track this new feature in a separate branch and keep you posted.

s3b3n, 2024-03-10:
Okay my fault i switched the power plug and the usb soundcard ports, the soundcard wasnt recognized, after switching erverything works for me

closed 2024-03-10 by nickpourazima