nickpourazima/rotary-phone-audio-guestbook #30: This repository is over its data quota

trackersos, 2024-03-18:
Hey,

I get following error:

`Downloading rpizero_rotary_phone_audio_guestbook_v1.0_imagebackup.img.gz (610 MB)
Error downloading object: rpizero_rotary_phone_audio_guestbook_v1.0_imagebackup.img.gz (ed056e0): Smudge error: Error downloading rpizero_rotary_phone_audio_guestbook_v1.0_imagebackup.img.gz (ed056e05beacf78a490eae1c0b5e5ebdac3cc340477580b259980830ac8e43c5): batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.`

Would you please check? Thank you very much!

nickpourazima, 2024-03-19:
Hey @trackersos, I just need to switch to GitHub releases so I'm not beholden to git lfs' strict quota. Will try to do this tomorrow morning and let you know.

nickpourazima, 2024-03-20:
@trackersos check out #32

closed 2024-03-20 by nickpourazima

trackersos, 2024-03-20:
> @trackersos check out #32 and the [gzipped img ](https://github.com/nickpourazima/rotary-phone-audio-guestbook/releases/download/v1.0.0/rpizero_rotary_phone_audio_guestbook_v1.0_imagebackup.img.gz)

@nickpourazima thank you, it works now!