nickpourazima/rotary-phone-audio-guestbook #35: Phone Headset Wiring

ChrishaJCG, 2024-05-05:
Hello, thank you for making this project, I am attempting to recreate it with the hardware that I have but I am running into an issue. I purchased a usb audio adapter but from the phone I will only ever get the speaker to be detected. Could you show the wiring from the handset jack to the usb audio adapter?

Thank you!

nickpourazima, 2024-05-05:
Hi @ChrishaJCG,

Sure. I took a few more pictures:

The USB audio adapter typically accepts 2 x 3.5mm male audio jacks which goes tip/ring/sleeve (TRS). I soldered my wires directly to the board but it isn't necessary for this project. This is what the top of my audio card looks like, note the differentiation between the microphone input and the speaker output:
[image]

In general a TRS jack maps to:
Tip (T) --> mic signal 1 (we'll call this positive M+)
Ring (R) --> mic signal 2 (we'll call this negative M-)
Sleeve (S) --> GND
[image]

And similarly for the speaker, T -> S+, R -> S-, S -> GND as in the picture:
[image]

Here is the mapping from the phone jack (left) to USB Audio Card (right):
Black --> Mic S+
Red --> Mic S- (or vis-versa, doesn't really matter)
Green --> GND
White --> Speaker (+)
And I left Speaker (-) disconnected since it wasn't necessary.

[image]

One thing to confirm from your side is whether your 3.5mm jack is TRS since it could be a TRRS which in that case an extra Ring is present (typically for headsets with built in microphones to carry all 3 signals (left audio, right audio, mic).

ChrishaJCG, 2024-05-09:
Hello! Thank you so much for these detailed pictures of the wiring. I was able to duplicate it and I am happy to report I am now getting audio out but also audio in on not only my machine but the PI that I am using as well.

closed 2024-05-14 by ChrishaJCG