nickpourazima/rotary-phone-audio-guestbook #36: Greeting_Start_Delay only accepts 0 or whole seconds, fractional seconds in decimal seems to introduce breaking error

Bonneykins, 2024-05-09:
https://github.com/nickpourazima/rotary-phone-audio-guestbook/blob/4c8bbd84d1a8fc324ebc9d6e4963807a303974e4/config.yaml#L19C1-L19C24

Tried to input a 1.5 second Greeting Start Delay and caused an error.

Could not find documentation from aplay about what is or isn't supported as duration flag parameter.

Just note this with in-line documentation for the meantime?

nickpourazima, 2024-05-15:
Thanks @Bonneykins for pointing that out! I'll look into it and keep you posted.

nickpourazima, 2024-05-20:
Hi @Bonneykins, I'm working on an update for this along with some other fixes, just need a bit more time after work to validate. Should be able to get it out this weekend.

PR #40 (2024-06-03): feat: address open issues, add recording limit, add fractional delay, add hangup stop
closes #27 
closes #29 
closes #36
closed 2024-06-03 by nickpourazima