nickpourazima/rotary-phone-audio-guestbook #42: Wrong file name on windows and android

Andys280, 2024-06-26:
I use this software successfully but I have a problem. In order to be able to extract the audio recordings, I installed a samba server on the raspberry pi. The problem is that in this server the files have totally strange names. I accessed via ssh the raspberry and in it the files have good names, date and time, but not on samba.
[image]
[image]

nickpourazima, 2024-06-26:
It looks like there might be a filename encoding issue with Samba. Try editing your Samba configuration file: `/etc/samba/smb.conf` on the Raspberry Pi. Then add/modify these parameters in the [global] section:
```
unix charset = UTF-8
dos charset = CP850
```
If that doesn't work, you might also need to add:
```
native language support = no
```
After making changes, restart the Samba service:
```
sudo systemctl restart smbd
```
This should align the character encoding between your Raspberry Pi and the Samba client. If you're still having problems, you might need to check the client-side settings as well.

Andys280, 2024-06-26:
same problem even with this lines added. It is possible to modify how the script create names ? I just want to number it 1,2,3,4.... not with date and time. Where can i change this ?

Andys280, 2024-06-26:
The weird thing is that filenames are good for sounds folder, i can see greeting.wav, beep.wav and everything else on samba. Problem appears only in recordings folder

nickpourazima, 2024-06-26:
Yes, in audioGuestBook.py just change this line:

```        
output_file = str(Path(self.config["recordings_path"]) / f"{datetime.now().isoformat()}.wav"
```

You can do something like:
```
count = 1
output_file = str(Path(self.config["recordings_path"]) / f"{count}.wav"
count++
```

closed 2024-06-28 by nickpourazima