nickpourazima/rotary-phone-audio-guestbook #61: Website Reload stops recording

s3b3n, 2024-09-16:


jmdevita, 2024-09-22:
hey @s3b3n, I was testing this with logs and I'm not able to reproduce the bug with my RPI zero 1w. I did notice however, that there is a delay in the recording being updated (~30 second delay between screen refresh and the length of the file reflecting a more accurate number).

Would you mind testing the below?
1) Start both services (audioGuestBook & Web Server)
2) Open up the webpage and pick up the phone to start the recording, record for ~20-30 seconds.
3) Refresh the page and wait 30 seconds. Then hang up the phone. Refresh the page again.
4) The website should say it was only recording for 20-30 seconds. However, wait ~30 seconds and refresh the page again. Does the recording time now reflect the new, longer timeframe?

digitaleJugend, 2025-01-19:
Hi ,

i use RPI Zero 1w with V 1.0.2. 

if i logged in over my PC with the Webserver/Website , i can`t see the "trash bin" Icon to delete files. (Firefox)
So I can`t delete Files over the Website. To listen to the files is possible. 

Also i can`t download the files as zip. from the website. The Zip is always empty  and have only 22kilobyte.
I only can transfer over ftp program.

[image]

nickpourazima, 2025-03-15:
check #70 and the latest [branch](https://github.com/nickpourazima/rotary-phone-audio-guestbook/tree/70-recording-non-stop-when-on-hook-within-greeting-delay-and-off-hook-again). I just pushed a potential fix for both of these issues.

nickpourazima, 2025-03-16:
Tested on Firefox

- Delete Icon Present:
[image]
- Download functional:
[image]

PR #72 (2025-03-16): 70 recording non stop when on hook within greeting delay and off hook again
Closes #61 
Closes #65
Closes #70
Closes #71
closed 2025-03-16 by nickpourazima