nickpourazima/rotary-phone-audio-guestbook #63: Add Reboot and shutdown button to website

s3b3n, 2024-09-16:
Reboot and shutdown the Pi over the web interface should be an option.

Thanks for the great work!

nickpourazima, 2024-09-16:
Shutdown is an option currently:

[image]

s3b3n, 2024-09-16:
> Shutdown is an option currently:
> 
> [image]
> 

Oh sry i missed that

nickpourazima, 2025-03-15:
added to #70 on [branch ](https://github.com/nickpourazima/rotary-phone-audio-guestbook/tree/70-recording-non-stop-when-on-hook-within-greeting-delay-and-off-hook-again)

will be bundling with next release

closed 2025-03-15 by nickpourazima

nickpourazima, 2025-03-16:
Added reboot and shutdown to top nav bar:

[image]