nickpourazima/rotary-phone-audio-guestbook #65: Gunicorn server crashes and restarts after wav file is played >30 seconds.

jmdevita, 2024-09-22:
I've noticed that when listening to a file back on the server, for example, a file longer than 30 seconds: 
[image]
The server will crash and restart after the 30 seconds is reached. From the client-side, there is no visible issue and the file can play to completion. But, the server will respond with this error message and restart.
[image]

Steps to Reproduce:
- Record or have a file >60 seconds in the recording folder and listen to it on the website.
- After you reach 30 seconds, refresh the page. You should see a "Internal System Error" message occur.
- After around 5-10 seconds, the server will be back online.

Current understanding:
- Gunicorn has a default worker timeout of 30 seconds. By clicking play on the audio file, the worker is running and not returning a response to the server. As a result, the server kills the worker after 30 seconds which causes the website to crash. I _think_ this may be due to the buffering of the file. If the file completely buffers, the worker will not be killed.

Non-Optimal Solution:
- We can increase the gunicorn worker timeout to a little over 5 minutes to resolve the issue (assuming the max recording time is 5 minutes), but when the recording is paused part way through and the 5 minute "timer" is reached, the server will crash.
`exec gunicorn -w 1 -b ${IP_ADDRESS}:8000 webserver.server:app --timeout 300`

nickpourazima, 2025-03-15:
Hi @jmdevita,

Appreciate the detailed report, and I think you diagnosed it perfectly. Rather than the partial fix of just increasing the timeout, I've attempted to implement a proper streaming solution that addresses the root cause:

- Added gevent to handle asynchronous workers:

```bash
exec gunicorn -w 1 -k gevent -b ${IP_ADDRESS}:8000 webserver.server:app
```

- Modified the server.py file to stream files in chunks rather than sending the entire file at once:

```python
@app.route("/recordings/<filename>")
def serve_recording(filename):
    """Serve a specific recording with proper streaming."""
    def generate():
        file_path = recordings_path / filename
        with open(file_path, 'rb') as f:
            chunk = f.read(4096)
            while chunk:
                yield chunk
                chunk = f.read(4096)
    
    return Response(
        generate(), 
        mimetype="audio/wav",
        headers={"Accept-Ranges": "bytes"}
    )
```

This way, the worker never times out because it's sending the file in small chunks rather than waiting for the entire file to be sent.
Could you pull the latest changes from [this branch](https://github.com/nickpourazima/rotary-phone-audio-guestbook/tree/70-recording-non-stop-when-on-hook-within-greeting-delay-and-off-hook-again) #70  and confirm whether this resolves the issue for you? The server should now handle long audio files without crashing, even if you pause/resume playback or leave the page open for extended periods.

Looking forward to your feedback!

nickpourazima, 2025-03-16:
Tested out a couple files >30 seconds with the new implementation and couldn't reproduce issue anymore. Closing w/ v1.0.3 release

PR #72 (2025-03-16): 70 recording non stop when on hook within greeting delay and off hook again
Closes #61 
Closes #65
Closes #70
Closes #71
closed 2025-03-16 by nickpourazima