nickpourazima/rotary-phone-audio-guestbook #66: Things stop working after a few recordings

mdavis1982, 2024-10-06:
I've just set up a new project with the latest release.

I'm using a normally closed switch. Everything appears to work fine for the first few recordings and then when putting the receiver down the recording just doesn't stop.

When this happens, toggling the switch a few more times sometimes makes things work again, but other times a reboot is required to stop the recording.

I'm not sure how to debug this.

nickpourazima, 2024-10-06:
Hi @mdavis1982 ,

Thanks for checking out this project!

Let's see if we can check a few things first:

1. Does an increase in `hook_bounce_time` resolve it?
   The current bounce time of 0.1 seconds may not be sufficient to filter out all spurious signals from the switch. Try increasing this value to 0.2 or 0.3 seconds:

   ```yaml
   hook_bounce_time: 0.3
   ```

2. Ensure that the switch is properly connected to the GPIO pin and ground. A loose connection could cause intermittent behavior.

3. If 1. doesn't address the situation we can talk about potentially adding an RC circuit:

[image]

And this would require a few software tweaks to dial in namely:

- Reduce the SW hook bounce time since utilizing the HW based RC debounce circuit:

```yaml
hook_bounce_time: 0.05
```

- Depending on how you've wired the circuit, you might need to invert the code logic. With the circuit I described above, the GPIO will read HIGH when the phone is off-hook and LOW when it's on-hook.

```python
hook = Button(22, pull_up=None, active_state=False)
```

Keep me posted on what you're seeing, I'm here to help!

mdavis1982, 2024-10-06:
@nickpourazima Thanks... I'll check this out,

Is there a way of tailing the logs so that I can see exactly what's happening as I'm running the project?

I'm using the release version just flashed to the SD card with Raspberry Pi Imager.

nickpourazima, 2024-10-06:
Check out the [debugging section](https://github.com/nickpourazima/rotary-phone-audio-guestbook?tab=readme-ov-file#debugging)

nickpourazima, 2025-03-15:
Hi @mdavis1982 , I'm working on a couple fixes for some other open issues. Can you let me know if this is still relevant based on the latest in the #70 [branch](https://github.com/nickpourazima/rotary-phone-audio-guestbook/tree/70-recording-non-stop-when-on-hook-within-greeting-delay-and-off-hook-again) and report in that thread? I'd like to consolidate fixes for this next release.

closed 2025-03-15 by nickpourazima