nickpourazima/rotary-phone-audio-guestbook #70: Recording non stop when ON hook within greeting delay and OFF hook again

ipanggggg, 2025-01-17:
Hi, 

Im using NC switch and everything works fine when we follow the proper flow: picking up the phone, listening to the greeting, leaving a message after the beep, and then placing the handset back. This time, I purposely tested what would happen if we repeatedly pick up and hang up the handset during the greeting because some of my sister wedding guests might play around and do this. 

I noticed that when we lift the handset and place it back before the greeting plays (On hook back within greeting_start_delay time), then lift it again while greeting is play, the recording will continue even though we have already placed the handset back. When we lift the handset again, the greeting will play as usual, but the recording is still from the previous one and when we place the handset back, the previous recording file is not separated and gets merged with the new greeting recording. 

I also noticed that when this happens, my USB audio card keeps blinking, indicating that it is in use, even though the handset has been placed back. Ideally, it should only blink when the handset is lifted and stop blinking when the handset is placed back, indicating that no recording is taking place. 

Im checking on `journalctl -fu audioGuestBook.service` there are no any error. 

> Jan 17 11:17:44 a13 python3[585]: INFO:__main__:Phone off hook, ready to begin!
> Jan 17 11:17:44 a13 python3[585]: INFO:__main__:Playing voicemail...
> Jan 17 11:17:44 a13 python3[856]: Simple mixer control 'Speaker',0
> Jan 17 11:17:44 a13 python3[856]:   Capabilities: pvolume pswitch pswitch-joined
> Jan 17 11:17:44 a13 python3[856]:   Playback channels: Front Left - Front Right
> Jan 17 11:17:44 a13 python3[856]:   Limits: Playback 0 - 151
> Jan 17 11:17:44 a13 python3[856]:   Mono:
> Jan 17 11:17:44 a13 python3[856]:   Front Left: Playback 151 [100%] [-0.06dB] [on]
> Jan 17 11:17:44 a13 python3[856]:   Front Right: Playback 151 [100%] [-0.06dB] [on]
> Jan 17 11:17:44 a13 python3[858]: Playing WAVE '/tmp/silence.wav' : Signed 32 bit Little Endian, Rate 44100 Hz, Stereo
> Jan 17 11:17:52 a13 python3[585]: INFO:__main__:Recording started...
> Jan 17 11:17:52 a13 python3[585]: INFO:__main__:Playing beep...
> Jan 17 11:17:53 a13 python3[862]: Simple mixer control 'Speaker',0
> Jan 17 11:17:53 a13 python3[862]:   Capabilities: pvolume pswitch pswitch-joined
> Jan 17 11:17:53 a13 python3[862]:   Playback channels: Front Left - Front Right
> Jan 17 11:17:53 a13 python3[862]:   Limits: Playback 0 - 151
> Jan 17 11:17:53 a13 python3[862]:   Mono:
> Jan 17 11:17:53 a13 python3[862]:   Front Left: Playback 151 [100%] [-0.06dB] [on]
> Jan 17 11:17:53 a13 python3[862]:   Front Right: Playback 151 [100%] [-0.06dB] [on]
> Jan 17 11:17:57 a13 python3[585]: INFO:__main__:Phone on hook. Ending call and saving recording.
> Jan 17 11:17:57 a13 python3[585]: INFO:audioInterface:Recording stopped.

the log mention `INFO:audioInterface:Recording stopped` but actually it not stopped when we off hook and on hook back within greeting delay and off hook back when greeting play. The recording will stop once reach recording limit or by rebooting the pi. 

i think this issue will happen when user off hook, on hook and off hook again within greeting delay time. 

Please assist, im very appreciate it

nickpourazima, 2025-01-17:
Thanks for testing this out @ipanggggg and grabbing the logs, much appreciated🙏🏽. I'll look into it soon.

nickpourazima, 2025-01-20:
@ipanggggg can you pull [this ](https://github.com/nickpourazima/rotary-phone-audio-guestbook/tree/70-recording-non-stop-when-on-hook-within-greeting-delay-and-off-hook-again)branch and test it out? I think I might have fixed it.

ipanggggg, 2025-01-20:
Hi @nickpourazima , will test it once i home and let you know. Thank You 🙏

nickpourazima, 2025-02-21:
Hey @ipanggggg, did this work out for you?

ipanggggg, 2025-02-23:
Hi @nickpourazima , Sorry for late respond 🙏🏻 i test it but the issue still persist. I dont face any issue regarding this when i use V1.0.0 and the issue arise when use V1.0.1 & V1.02.. Maybe it was due stop playing greeting when guest on-hook the phone while greeting message playing?

nickpourazima, 2025-03-15:
@ipanggggg no worries, sorry I haven't had as much time to spend on this lately. 

I've tried to implement a fix which will make sure:

- When you put the phone down, all recordings stop properly every time
- The system does a better job of cleaning up after itself between calls
- If somehow a recording is still running when it shouldn't be, there's now a backup method to stop it

Can you let me know if it works out for you?

PR #72 (2025-03-16): 70 recording non stop when on hook within greeting delay and off hook again
Closes #61 
Closes #65
Closes #70
Closes #71
closed 2025-03-16 by nickpourazima