nickpourazima/rotary-phone-audio-guestbook #71: probl with saving records

quba0815, 2025-03-13:
Hi. I am trying to use this project. sound is playing well but it is not recording.

I see in journal

Mar 13 23:38:42 rpizero-rotary-phone-audio-guestbook python3[823]:   Front Right: Playback 37 [100%] [0.00dB] [on]
Mar 13 23:38:43 rpizero-rotary-phone-audio-guestbook python3[825]: Playing WAVE '/tmp/silence.wav' : Signed 32 bit Little Endian, Rate 44100 Hz, Stereo
Mar 13 23:38:43 rpizero-rotary-phone-audio-guestbook python3[825]: aplay: set_params:1343: Sample format non available
Mar 13 23:38:43 rpizero-rotary-phone-audio-guestbook python3[825]: Available formats:
Mar 13 23:38:43 rpizero-rotary-phone-audio-guestbook python3[825]: - S16_LE
Mar 13 23:38:43 rpizero-rotary-phone-audio-guestbook python3[546]: ERROR:audioInterface:Error generating or playing silence file: Command '['aplay', '-D', 'hw:1,0', '/tmp/silence.wav']' returned non-zero exit status 1.
Mar 13 23:38:45 rpizero-rotary-phone-audio-guestbook python3[546]: INFO:__main__:Phone on hook. Ending call and saving recording.
Mar 13 23:38:45 rpizero-rotary-phone-audio-guestbook python3[546]: INFO:audioInterface:Recording stopped.
Mar 13 23:38:45 rpizero-rotary-phone-audio-guestbook python3[546]: INFO:__main__:Stopping playback.
Mar 13 23:38:46 rpizero-rotary-phone-audio-guestbook python3[546]: INFO:audioInterface:Playback stopped.
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[546]: INFO:__main__:Phone off hook, ready to begin!
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[546]: INFO:__main__:Playing voicemail...
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[892]: Simple mixer control 'Speaker',0
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[892]:   Capabilities: pvolume pswitch pswitch-joined
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[892]:   Playback channels: Front Left - Front Right
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[892]:   Limits: Playback 0 - 37
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[892]:   Mono:
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[892]:   Front Left: Playback 37 [100%] [0.00dB] [on]
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[892]:   Front Right: Playback 37 [100%] [0.00dB] [on]
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[894]: Playing WAVE '/tmp/silence.wav' : Signed 32 bit Little Endian, Rate 44100 Hz, Stereo
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[894]: aplay: set_params:1343: Sample format non available
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[894]: Available formats:
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[894]: - S16_LE
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[546]: ERROR:audioInterface:Error generating or playing silence file: Command '['aplay', '-D', 'hw:1,0', '/tmp/silence.wav']' returned non-zero exit status 1.
Mar 13 23:49:24 rpizero-rotary-phone-audio-guestbook python3[546]: INFO:__main__:Recording started...
Mar 13 23:49:24 rpizero-rotary-phone-audio-guestbook python3[546]: INFO:__main__:Playing beep...
Mar 13 23:49:24 rpizero-rotary-phone-audio-guestbook python3[898]: Simple mixer control 'Speaker',0
Mar 13 23:49:24 rpizero-rotary-phone-audio-guestbook python3[898]:   Capabilities: pvolume pswitch pswitch-joined
Mar 13 23:49:24 rpizero-rotary-phone-audio-guestbook python3[898]:   Playback channels: Front Left - Front Right
Mar 13 23:49:24 rpizero-rotary-phone-audio-guestbook python3[898]:   Limits: Playback 0 - 37
Mar 13 23:49:24 rpizero-rotary-phone-audio-guestbook python3[898]:   Mono:
Mar 13 23:49:24 rpizero-rotary-phone-audio-guestbook python3[898]:   Front Left: Playback 37 [100%] [0.00dB] [on]
Mar 13 23:49:24 rpizero-rotary-phone-audio-guestbook python3[898]:   Front Right: Playback 37 [100%] [0.00dB] [on]
Mar 13 23:49:35 rpizero-rotary-phone-audio-guestbook python3[546]: INFO:__main__:Phone on hook. Ending call and saving recording.
Mar 13 23:49:35 rpizero-rotary-phone-audio-guestbook python3[546]: INFO:audioInterface:Recording stopped.

But no recording saved.

Also if I try in the webserver to change settings nothing happens when I click the "Save" button. Only possible to change settings directly in the config.yaml file

could you help me out?
If possible I wanted to use it on weekend at a party ;-)

Many Thanks

nickpourazima, 2025-03-14:
Hi @quba0815, thanks for the write-up! I'll take a look and get back to you soon.

nickpourazima, 2025-03-15:
@quba0815 

One thing I noticed checking out the logs you sent:
```bash
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[894]: Playing WAVE '/tmp/silence.wav' : Signed 32 bit Little Endian, Rate 44100 Hz, Stereo
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[894]: Available formats:
Mar 13 23:49:17 rpizero-rotary-phone-audio-guestbook python3[894]: - S16_LE
```

This is saying that your audio hardware only support the S16_LE (signed 16-bit little endian format) but the system is trying to use the default format specified in the `config.yaml`.

Can you edit your `config.yaml` to change the format from `cd` to `S16_LE`?

It should look like:
```bash
format: S16_LE
```

SSH into your rpi and execute something like:
```bash
sudo nano /home/admin/rotary-phone-audio-guestbook/config.yaml
```

Find the format line and change it, then save with Ctrl+O, Enter, and exit with Ctrl+X.

nickpourazima, 2025-03-15:
For the webserver Save button issue, I think I fixed it in [this branch](https://github.com/nickpourazima/rotary-phone-audio-guestbook/tree/70-recording-non-stop-when-on-hook-within-greeting-delay-and-off-hook-again) from #70:

Can you:
- `git checkout 70-recording-non-stop-when-on-hook-within-greeting-delay-and-off-hook-again`
- fix your config.yaml with the proper format mentioned above
- `sudo systemctl restart audioGuestBook.service`

nickpourazima, 2025-03-16:
Confirmed fixed in v1.0.3

PR #72 (2025-03-16): 70 recording non stop when on hook within greeting delay and off hook again
Closes #61 
Closes #65
Closes #70
Closes #71
closed 2025-03-16 by nickpourazima