nickpourazima/rotary-phone-audio-guestbook #75: NO configuration doesnt start the playback

trfr3ak, 2025-03-29:
I have a strange problem with my setup of the audio guestbook.
My phone has a NO hook, but it only starts recording when using the NC config (but then of course in a reversed behaviour, as it starts recording when putting the phone down)
Is this a known bug or am I doing something wrong?
Thank you very much for your great project and documentation,  as it helped me already a lot for the audio config!

nickpourazima, 2025-03-29:
Hey @trfr3ak! You're very welcome, glad I could help! As far as the NO vs. NC, just wanted to confirm that you've tried to modify the config.yaml to NO (either on the webpage or manually editing the file in the path) and either reboot your pi or `sudo systemctl restart audioGuestbook.service`? If so, then I can maybe narrow down a couple other potential probable causes here.

trfr3ak, 2025-03-29:
Hey @nickpourazima ,
thanks for the quick replay.
Yes I already checked it. I always have to restart the service in Order for any change to be activated, as the webinterface doesnt restart the service right?

trfr3ak, 2025-03-29:
I found another issue with my setup, which seems to be connected to the issue.
When I'm using the NC configuration for testing I hear the message and the beep, but it never stops recording, even after pushing the button, so it seems like it is not recognizing one state of the hook right? 
The log with the "reversed" NC looks like this:
`
-- Journal begins at Sat 2025-03-15 21:44:34 CET. --
Mar 29 21:35:08 audioguestbook python3[797]:   Front Right: Playback 37 [100%] [0.00dB] [on]
Mar 29 21:35:19 audioguestbook python3[784]: INFO:__main__:Phone on hook. Ending call and saving recording.
Mar 29 21:35:19 audioguestbook systemd[1]: audioGuestBook.service: Succeeded.
Mar 29 21:35:20 audioguestbook systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 1.
Mar 29 21:35:20 audioguestbook systemd[1]: Stopped Rotary Phone Guest Book Project.
Mar 29 21:35:20 audioguestbook systemd[1]: Started Rotary Phone Guest Book Project.
Mar 29 21:35:20 audioguestbook python3[799]: INFO:__main__:Using configuration file: /home/admin/rotary-phone-audio-guestbook/src/../conf                                        ig.yaml
Mar 29 21:35:20 audioguestbook python3[799]: INFO:__main__:record_greeting_gpio is 0, skipping setup.
Mar 29 21:35:20 audioguestbook python3[799]: INFO:__main__:no shutdown button declared, skipping button init
Mar 29 21:35:20 audioguestbook python3[799]: INFO:__main__:System ready. Lift the handset to start.
Mar 29 21:36:41 audioguestbook python3[799]: INFO:__main__:Phone off hook, ready to begin!
Mar 29 21:36:41 audioguestbook python3[799]: INFO:__main__:Playing voicemail...
Mar 29 21:36:41 audioguestbook python3[808]: Simple mixer control 'Speaker',0
Mar 29 21:36:41 audioguestbook python3[808]:   Capabilities: pvolume pswitch pswitch-joined
Mar 29 21:36:41 audioguestbook python3[808]:   Playback channels: Front Left - Front Right
Mar 29 21:36:41 audioguestbook python3[808]:   Limits: Playback 0 - 37
Mar 29 21:36:41 audioguestbook python3[808]:   Mono:
Mar 29 21:36:41 audioguestbook python3[808]:   Front Left: Playback 37 [100%] [0.00dB] [on]
Mar 29 21:36:41 audioguestbook python3[808]:   Front Right: Playback 37 [100%] [0.00dB] [on]
Mar 29 21:36:41 audioguestbook python3[810]: Playing WAVE '/tmp/silence.wav' : Signed 32 bit Little Endian, Rate 44100 Hz, Stereo
Mar 29 21:36:49 audioguestbook python3[799]: INFO:__main__:Recording started...
Mar 29 21:36:49 audioguestbook python3[799]: INFO:__main__:Playing beep...
Mar 29 21:36:49 audioguestbook python3[814]: Simple mixer control 'Speaker',0
Mar 29 21:36:49 audioguestbook python3[814]:   Capabilities: pvolume pswitch pswitch-joined
Mar 29 21:36:49 audioguestbook python3[814]:   Playback channels: Front Left - Front Right
Mar 29 21:36:49 audioguestbook python3[814]:   Limits: Playback 0 - 37
Mar 29 21:36:49 audioguestbook python3[814]:   Mono:
Mar 29 21:36:49 audioguestbook python3[814]:   Front Left: Playback 37 [100%] [0.00dB] [on]
Mar 29 21:36:49 audioguestbook python3[814]:   Front Right: Playback 37 [100%] [0.00dB] [on]
Mar 29 21:41:49 audioguestbook python3[799]: INFO:__main__:Recording time exceeded. Stopping recording.
Mar 29 21:41:49 audioguestbook systemd[1]: audioGuestBook.service: Succeeded.
Mar 29 21:41:49 audioguestbook systemd[1]: audioGuestBook.service: Consumed 1.251s CPU time.
Mar 29 21:41:49 audioguestbook systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 2.
Mar 29 21:41:49 audioguestbook systemd[1]: Stopped Rotary Phone Guest Book Project.
Mar 29 21:41:49 audioguestbook systemd[1]: audioGuestBook.service: Consumed 1.251s CPU time.
Mar 29 21:41:49 audioguestbook systemd[1]: Started Rotary Phone Guest Book Project.
Mar 29 21:41:50 audioguestbook python3[857]: INFO:__main__:Using configuration file: /home/admin/rotary-phone-audio-guestbook/src/../conf                                        ig.yaml
Mar 29 21:41:50 audioguestbook python3[857]: INFO:__main__:record_greeting_gpio is 0, skipping setup.
Mar 29 21:41:50 audioguestbook python3[857]: INFO:__main__:no shutdown button declared, skipping button init
Mar 29 21:41:50 audioguestbook python3[857]: INFO:__main__:System ready. Lift the handset to start.
`

nickpourazima, 2025-03-29:
Hey @trfr3ak, I believe this is occurring because different phone models can have different electrical characteristics for their hook switches, which can lead to reversed behavior like you described - where your NO hook only works in NC mode, but with the recording starting when you put the phone down.

I've added a new configuration option called `invert_hook` that should solve this problem. To test it, can you either:
- pull this branch on your pi

**OR**

- manually copy/paste the changes (namely the `*.py` and `*.html`), then restart the systemctl services with `sudo systemctl restart audioGuestBook.service`

New `config.yaml` settings in this branch:
```markdown
hook_type: NO (since your phone has a NO hook as you mentioned)
invert_hook: true (this is a new setting I've added)
```

This configuration will keep your correct physical hook type (NO) while fixing the reversed behavior. The phone should now record when lifting the handset and stop when putting it down.

I've also updated the web interface to include this option and added documentation on troubleshooting hook configuration issues. 
Also, the configuration will now automatically restart the service when changes are made through the web interface - thanks for pointing that out!

Could you please try this fix and let me know if it resolves your issue?

trfr3ak, 2025-03-29:
Hey @nickpourazima!
now we are getting closer! Unfortunately the behaviour with NO setting is still the same, but when using the NC, it starts the message when lifting the earpiece up, and it stops the recording when putting it down!
One last thing that doesnt seem to work is the recording,
the Log says Ending call and saving recording, but it is nowhere to be found. Is there any way to debug this further? Because I dont see an error message. 
arecord works fine btw

nickpourazima, 2025-03-29:
Okay, I've just pushed a simpler hook mapping logic and added extensive debugging to track down the recording issue.

For the hook behavior:
- Good to hear NC + invert_hook works! That confirms our fix is on the right track.
- For NO setting, check the logs for "Hook setup:" and "Button initialized with" messages. It's possible there's a hardware-specific behavior with your particular phone model.

For the missing recordings:
- Check the logs for "Will save recording to: /path/to/file.wav" messages to confirm the exact path
- Verify permissions: `sudo chown -R admin:admin /home/admin/rotary-phone-audio-guestbook/recordings`
- Look for partial files - sometimes recordings are created but not properly finalized
- The latest code includes detailed logging for the arecord process - look for any stderr output from arecord

Let me know what you find in the debug logs. The fix for the hook is working for some configurations, so we're getting closer to solving both issues!

trfr3ak, 2025-03-29:
Unfortunately the log files arent mentioning your line with "will save recording." This is the complete log after one recording cycle with the phone:
```

-- Journal begins at Sat 2025-03-15 21:44:34 CET. --
Mar 30 00:47:07 audioguestbook python3[1232]:   Front Right: Playback 37 [100%] [0.00dB] [on]
Mar 30 00:47:19 audioguestbook python3[1219]: INFO:__main__:Phone on hook. Ending call and saving recording.
Mar 30 00:47:19 audioguestbook systemd[1]: audioGuestBook.service: Succeeded.
Mar 30 00:47:19 audioguestbook systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 1.
Mar 30 00:47:19 audioguestbook systemd[1]: Stopped Rotary Phone Guest Book Project.
Mar 30 00:47:19 audioguestbook systemd[1]: Started Rotary Phone Guest Book Project.
Mar 30 00:47:19 audioguestbook python3[1235]: INFO:__main__:Using configuration file: /home/admin/rotary-phone-audio-guestbook/src/../config.yaml
Mar 30 00:47:20 audioguestbook python3[1235]: INFO:__main__:record_greeting_gpio is 0, skipping setup.
Mar 30 00:47:20 audioguestbook python3[1235]: INFO:__main__:no shutdown button declared, skipping button init
Mar 30 00:47:20 audioguestbook python3[1235]: INFO:__main__:System ready. Lift the handset to start.
Mar 30 00:54:12 audioguestbook python3[1235]: INFO:__main__:Phone off hook, ready to begin!
Mar 30 00:54:12 audioguestbook python3[1235]: INFO:__main__:Playing voicemail...
Mar 30 00:54:12 audioguestbook python3[1258]: Simple mixer control 'Speaker',0
Mar 30 00:54:12 audioguestbook python3[1258]:   Capabilities: pvolume pswitch pswitch-joined
Mar 30 00:54:12 audioguestbook python3[1258]:   Playback channels: Front Left - Front Right
Mar 30 00:54:12 audioguestbook python3[1258]:   Limits: Playback 0 - 37
Mar 30 00:54:12 audioguestbook python3[1258]:   Mono:
Mar 30 00:54:12 audioguestbook python3[1258]:   Front Left: Playback 37 [100%] [0.00dB] [on]
Mar 30 00:54:12 audioguestbook python3[1258]:   Front Right: Playback 37 [100%] [0.00dB] [on]
Mar 30 00:54:12 audioguestbook python3[1260]: Playing WAVE '/tmp/silence.wav' : Signed 32 bit Little Endian, Rate 44100 Hz, Stereo
Mar 30 00:54:20 audioguestbook python3[1235]: INFO:__main__:Recording started...
Mar 30 00:54:20 audioguestbook python3[1235]: INFO:__main__:Playing beep...
Mar 30 00:54:21 audioguestbook python3[1264]: Simple mixer control 'Speaker',0
Mar 30 00:54:21 audioguestbook python3[1264]:   Capabilities: pvolume pswitch pswitch-joined
Mar 30 00:54:21 audioguestbook python3[1264]:   Playback channels: Front Left - Front Right
Mar 30 00:54:21 audioguestbook python3[1264]:   Limits: Playback 0 - 37
Mar 30 00:54:21 audioguestbook python3[1264]:   Mono:
Mar 30 00:54:21 audioguestbook python3[1264]:   Front Left: Playback 37 [100%] [0.00dB] [on]
Mar 30 00:54:21 audioguestbook python3[1264]:   Front Right: Playback 37 [100%] [0.00dB] [on]
Mar 30 00:54:28 audioguestbook python3[1235]: INFO:__main__:Phone on hook. Ending call and saving recording.
Mar 30 00:54:28 audioguestbook systemd[1]: audioGuestBook.service: Succeeded.
Mar 30 00:54:28 audioguestbook systemd[1]: audioGuestBook.service: Consumed 1.290s CPU time.
Mar 30 00:54:28 audioguestbook systemd[1]: audioGuestBook.service: Scheduled restart job, restart counter is at 2.
Mar 30 00:54:28 audioguestbook systemd[1]: Stopped Rotary Phone Guest Book Project.
Mar 30 00:54:28 audioguestbook systemd[1]: audioGuestBook.service: Consumed 1.290s CPU time.
Mar 30 00:54:28 audioguestbook systemd[1]: Started Rotary Phone Guest Book Project.
Mar 30 00:54:29 audioguestbook python3[1266]: INFO:__main__:Using configuration file: /home/admin/rotary-phone-audio-guestbook/src/../config.yaml
Mar 30 00:54:29 audioguestbook python3[1266]: INFO:__main__:record_greeting_gpio is 0, skipping setup.
Mar 30 00:54:29 audioguestbook python3[1266]: INFO:__main__:no shutdown button declared, skipping button init
Mar 30 00:54:29 audioguestbook python3[1266]: INFO:__main__:System ready. Lift the handset to start.

```

The file permission were correct. I didnt find any partial files.
I didnt quit find the logs you are mentioning here: 

> The latest code includes detailed logging for the arecord process - look for any stderr output from arecord

trfr3ak, 2025-03-29:
Sorry, was too quick to respond, now I see the new commit! Will check it out, thank you very much!

trfr3ak, 2025-03-30:
Okay, now I get an error message. Something with the arecord seems to be wrong:
```
Mar 30 01:09:49 audioguestbook python3[1334]: INFO:audioInterface:Starting recording to: /home/admin/rotary-phone-audio-guestbook/recordings/2025-03-30T01:09:49.994588.wav
Mar 30 01:09:49 audioguestbook python3[1334]: INFO:audioInterface:Recording command: arecord -D plughw:1,0 -f cd -t wav  -d 300 -r 44100 -c 2 /home/admin/rotary-phone-audio-guestbook/recordings/2025-03-30T01:09:49.994588.wav
Mar 30 01:09:50 audioguestbook python3[1334]: INFO:audioInterface:Recording process started with PID: 1344
Mar 30 01:09:50 audioguestbook python3[1334]: INFO:__main__:Recording started...
Mar 30 01:09:50 audioguestbook python3[1334]: INFO:__main__:Playing beep...
Mar 30 01:09:50 audioguestbook python3[1346]: Simple mixer control 'Speaker',0
Mar 30 01:09:50 audioguestbook python3[1346]:   Capabilities: pvolume pswitch pswitch-joined
Mar 30 01:09:50 audioguestbook python3[1346]:   Playback channels: Front Left - Front Right
Mar 30 01:09:50 audioguestbook python3[1346]:   Limits: Playback 0 - 37
Mar 30 01:09:50 audioguestbook python3[1346]:   Mono:
Mar 30 01:09:50 audioguestbook python3[1346]:   Front Left: Playback 37 [100%] [0.00dB] [on]
Mar 30 01:09:50 audioguestbook python3[1346]:   Front Right: Playback 37 [100%] [0.00dB] [on]
Mar 30 01:10:01 audioguestbook python3[1334]: INFO:__main__:Phone on hook. Ending call and saving recording.
Mar 30 01:10:01 audioguestbook python3[1334]: INFO:audioInterface:Stopping recording process with PID: 1344
Mar 30 01:10:01 audioguestbook python3[1334]: INFO:audioInterface:arecord stderr output: arecord: main:599: unrecognized file format wav
Mar 30 01:10:01 audioguestbook python3[1334]: WARNING:audioInterface:Process termination issue: [Errno 3] No such process, sending SIGKILL
Mar 30 01:10:01 audioguestbook python3[1334]: ERROR:audioInterface:Error stopping recording process: local variable 'pgid' referenced before assignment
Mar 30 01:10:01 audioguestbook python3[1334]: INFO:audioInterface:Recording stopped

```

nickpourazima, 2025-03-30:
Okay but `arecord` by itself was working for you, correct? Something like:

`arecord -D plughw:1,0 -f cd -t wav -d 10 -r 44100 -c 2 /home/admin/rotary-phone-audio-guestbook/recordings/test.wav`

trfr3ak, 2025-03-30:
Yes, this is working fine and I can listen to the Audio now on the Webinterface

nickpourazima, 2025-03-30:
Okay I think there are two specific problems causing the recordings to fail:

1. The `arecord` command is failing with "unrecognized file format wav" - this indicates an issue with how the parameters are being parsed. I've fixed this by ensuring the `file_type` parameter is properly cleaned of any whitespace and properly formatted in the command.

2. The process termination is failing with a "variable referenced before assignment" error, which means our error handling isn't properly handling cases where the process might already be terminated before we try to kill it.

Could you pull the latest commit I just pushed and test if recordings are now being saved properly?

trfr3ak, 2025-03-30:
You're a hero man! Recording works now!
Only issue now is, that it isnt restarting the AudioGuestBook.py after saving the recording, maybe it got lost on the way on the last commits?

nickpourazima, 2025-03-30:
Can you explain a little further? Do you mean that it's not continuing to work after the first recording?

nickpourazima, 2025-03-30:
I pushed one more commit, refactoring a bit of the recording processing logic, let me know if it solves your issue!

trfr3ak, 2025-03-30:
Yes, it was stuck on stopped recording, but didnt reset for new recording, whcih it does now with your new commit.
Unfortunately now the files arent saved anymore again, here is the log:

```

Mar 30 10:46:19 audioguestbook python3[780]: INFO:__main__:Verifying event listeners are active
Mar 30 10:46:19 audioguestbook python3[780]: INFO:__main__:=========================================
Mar 30 10:46:19 audioguestbook python3[780]: INFO:__main__:System reset completed successfully
Mar 30 10:46:19 audioguestbook python3[780]: INFO:__main__:Ready for next recording - lift handset to begin
Mar 30 10:46:19 audioguestbook python3[780]: INFO:__main__:=========================================
Mar 30 10:47:40 audioguestbook python3[780]: INFO:__main__:Phone off hook, ready to begin!
Mar 30 10:47:40 audioguestbook python3[780]: INFO:audioInterface:No active recording process to stop, checking for strays
Mar 30 10:47:40 audioguestbook python3[780]: INFO:__main__:Verifying event listeners are active
Mar 30 10:47:40 audioguestbook python3[780]: INFO:__main__:Playing voicemail...
Mar 30 10:47:40 audioguestbook python3[806]: Simple mixer control 'Speaker',0
Mar 30 10:47:40 audioguestbook python3[806]:   Capabilities: pvolume pswitch pswitch-joined
Mar 30 10:47:40 audioguestbook python3[806]:   Playback channels: Front Left - Front Right
Mar 30 10:47:40 audioguestbook python3[806]:   Limits: Playback 0 - 37
Mar 30 10:47:40 audioguestbook python3[806]:   Mono:
Mar 30 10:47:40 audioguestbook python3[806]:   Front Left: Playback 37 [100%] [0.00dB] [on]
Mar 30 10:47:40 audioguestbook python3[806]:   Front Right: Playback 37 [100%] [0.00dB] [on]
Mar 30 10:47:40 audioguestbook python3[808]: Playing WAVE '/tmp/silence.wav' : Signed 32 bit Little Endian, Rate 44100 Hz, Stereo
Mar 30 10:47:49 audioguestbook python3[780]: INFO:__main__:Will save recording to: /home/admin/rotary-phone-audio-guestbook/recordings/2025-03-30T10:47:49.309693.wav
Mar 30 10:47:49 audioguestbook python3[780]: INFO:__main__:Recording directory exists: True
Mar 30 10:47:49 audioguestbook python3[780]: INFO:__main__:Recording directory is writable: True
Mar 30 10:47:49 audioguestbook python3[780]: INFO:audioInterface:Started recording process with PID: 811
Mar 30 10:47:49 audioguestbook python3[780]: INFO:__main__:Recording started...
Mar 30 10:47:49 audioguestbook python3[780]: INFO:__main__:Playing beep...
Mar 30 10:47:49 audioguestbook python3[813]: Simple mixer control 'Speaker',0
Mar 30 10:47:49 audioguestbook python3[813]:   Capabilities: pvolume pswitch pswitch-joined
Mar 30 10:47:49 audioguestbook python3[813]:   Playback channels: Front Left - Front Right
Mar 30 10:47:49 audioguestbook python3[813]:   Limits: Playback 0 - 37
Mar 30 10:47:49 audioguestbook python3[813]:   Mono:
Mar 30 10:47:49 audioguestbook python3[813]:   Front Left: Playback 37 [100%] [0.00dB] [on]
Mar 30 10:47:49 audioguestbook python3[813]:   Front Right: Playback 37 [100%] [0.00dB] [on]
Mar 30 10:48:02 audioguestbook python3[780]: INFO:__main__:Phone on hook. Ending call and saving recording.
Mar 30 10:48:02 audioguestbook python3[780]: INFO:audioInterface:Stopping recording process with PID: 811
Mar 30 10:48:02 audioguestbook python3[780]: INFO:audioInterface:Recording process terminated gracefully
Mar 30 10:48:02 audioguestbook python3[780]: INFO:audioInterface:Recording stopped and process tracking reset
Mar 30 10:48:02 audioguestbook python3[780]: INFO:__main__:Verifying event listeners are active
Mar 30 10:48:02 audioguestbook python3[780]: INFO:__main__:=========================================
Mar 30 10:48:02 audioguestbook python3[780]: INFO:__main__:System reset completed successfully
Mar 30 10:48:02 audioguestbook python3[780]: INFO:__main__:Ready for next recording - lift handset to begin
Mar 30 10:48:02 audioguestbook python3[780]: INFO:__main__:=========================================

```

nickpourazima, 2025-03-30:
Okay, I think I know what might be going on - just pushed the last commit (hopefully). Thanks for your patience and for working through the issues w/ me!

trfr3ak, 2025-03-30:
Nice! No its working fine! Thank you so much!

I found a weird thing, which only happens if you tap the hook quickly a few times (because its not always responding, maybe I have to change to another switch on the inside, because it seems a weird switch..). But after doing the mention thing I get this error message and the script is stuck and I have to restart it manually in order to get it working again:
```

Mar 30 14:02:12 audioguestbook python3[749]: INFO:__main__:Verifying event listeners are active
Mar 30 14:02:12 audioguestbook python3[749]: INFO:__main__:=========================================
Mar 30 14:02:12 audioguestbook python3[749]: INFO:__main__:System reset completed successfully
Mar 30 14:02:12 audioguestbook python3[749]: INFO:__main__:Ready for next recording - lift handset to begin
Mar 30 14:02:12 audioguestbook python3[749]: INFO:__main__:=========================================
Mar 30 14:02:44 audioguestbook python3[749]: INFO:__main__:Phone off hook, ready to begin!
Mar 30 14:02:44 audioguestbook python3[749]: INFO:audioInterface:Stopping recording process with PID: 777
Mar 30 14:02:44 audioguestbook python3[749]: INFO:audioInterface:Recording process terminated gracefully
Mar 30 14:02:44 audioguestbook python3[749]: WARNING:audioInterface:Recording file not found or empty: /home/admin/rotary-phone-audio-guestbook/recordings/2025-03-30T14:02:12.744070.wav
Mar 30 14:02:45 audioguestbook python3[749]: INFO:audioInterface:Recording stopped and process tracking reset
Mar 30 14:02:45 audioguestbook python3[749]: Traceback (most recent call last):
Mar 30 14:02:45 audioguestbook python3[749]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 223, in _call_when_changed
Mar 30 14:02:45 audioguestbook python3[749]:     super(RPiGPIOPin, self)._call_when_changed()
Mar 30 14:02:45 audioguestbook python3[749]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 128, in _call_when_changed
Mar 30 14:02:45 audioguestbook python3[749]:     super(LocalPiPin, self)._call_when_changed(
Mar 30 14:02:45 audioguestbook python3[749]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 298, in _call_when_changed
Mar 30 14:02:45 audioguestbook python3[749]:     method(ticks, state)
Mar 30 14:02:45 audioguestbook python3[749]:   File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 182, in _pin_changed
Mar 30 14:02:45 audioguestbook python3[749]:     self._fire_events(ticks, bool(self._state_to_value(state)))
Mar 30 14:02:45 audioguestbook python3[749]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 405, in _fire_events
Mar 30 14:02:45 audioguestbook python3[749]:     self._fire_deactivated()
Mar 30 14:02:45 audioguestbook python3[749]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 452, in _fire_deactivated
Mar 30 14:02:45 audioguestbook python3[749]:     super(HoldMixin, self)._fire_deactivated()
Mar 30 14:02:45 audioguestbook python3[749]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 369, in _fire_deactivated
Mar 30 14:02:45 audioguestbook python3[749]:     self.when_deactivated()
Mar 30 14:02:45 audioguestbook python3[749]:   File "/home/admin/rotary-phone-audio-guestbook/src/audioGuestBook.py", line 147, in off_hook
Mar 30 14:02:45 audioguestbook python3[749]:     self.stop_recording_and_playback()
Mar 30 14:02:45 audioguestbook python3[749]:   File "/home/admin/rotary-phone-audio-guestbook/src/audioGuestBook.py", line 337, in stop_recording_and_playback
Mar 30 14:02:45 audioguestbook python3[749]:     if hasattr(self, "greeting_thread") and self.greeting_thread.is_alive():
Mar 30 14:02:45 audioguestbook python3[749]: AttributeError: 'NoneType' object has no attribute 'is_alive'

```

nickpourazima, 2025-03-30:
I see the problem - it's a race condition that happens when the hook is toggled rapidly.

I've just pushed a fix that adds better error handling in the thread management code. This should prevent the system from crashing even if the hook switch is toggled rapidly or behaves erratically.

If you're having issues with the switch itself, you might want to look into adding some additional [hardware debouncing](https://raspberrypi.stackexchange.com/questions/118349/what-is-the-proper-way-to-debounce-a-gpio-input) or increasing the software debounce time in the `config.yaml` (currently set to 0.1 seconds). A value of 0.2 or 0.3 might help with an unreliable switch.

nickpourazima, 2025-03-30:
Should I close this out now? I was planning to squash these commits and build a new release for these fixes. Again, I apprecaite your help troubleshooting!

trfr3ak, 2025-03-30:
Thanks so much!
Yes I tried the software debouncing from values 0.04 to 0.8, but none of the values was working 100% fine.
So I'll look at the hardware debouncing thanks for the idea!
And thank you so much for everything!

nickpourazima, 2025-03-30:
You're very welcome! Hmm, yeah it that case a simple low pass filter (RC circuit) might do the trick, or if you don't have the parts laying around you could also try to add a little more weight to the actual handset? That might be what it needs in order to definitively depress the hook.

commit 5f7a5a14 (2025-03-30): Fix audio recording and hook detection issues

- Fix hook detection logic to properly handle all combinations of NC/NO switches and inversion settings
- Improve audio file saving by using SIGINT for graceful process termination
- Add proper process group management for clean recording process handling
- Implement stepped termination approach (SIGINT → SIGTERM → SIGKILL) with appropriate delays
- Add file verification to confirm recordings are properly saved
- Add directory existence and permission checks before recording
- Improve error handling and logging throughout the code

Resolves #75