nickpourazima/rotary-phone-audio-guestbook #79: Host AP instead of having to connect to another network

nickpourazima, 2025-04-20:
### Discussed in https://github.com/nickpourazima/rotary-phone-audio-guestbook/discussions/78

<div type='discussions-op-text'>

<sup>Originally posted by **mlcrow** April 16, 2025</sup>
I think it'd be really useful if the pi booted to host a wireless access point rather than having to configure it to connect to a predefined wireless network. I'm not super well versed in this but if you think it's possible, I think it'd be a useful feature.

Thanks for your hard work and great documentation!</div>

CoachTacamo, 2025-07-14:
I've succeeded setting this up!

```
sudo apt update
sudo apt install hostapd dnsmasq

sudo systemctl stop hostapd
sudo systemctl stop dnsmasq

sudo nano /etc/dhcpcd.conf

interface wlan0
    static ip_address=192.168.4.1/24
    nohook wpa_supplicant

sudo mv /etc/dnsmasq.conf /etc/dnsmasq.conf.orig

interface=wlan0
dhcp-range=192.168.4.2,192.168.4.20,255.255.255.0,24h

sudo nano /etc/hostapd/hostapd.conf

interface=wlan0
driver=nl80211
ssid=Pi_AP
hw_mode=g
channel=7
wmm_enabled=0
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=raspberry123
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

sudo nano /etc/default/hostapd
DAEMON_CONF="/etc/hostapd/hostapd.conf"
sudo nano /etc/sysctl.conf
net.ipv4.ip_forward=1
sudo sh -c "echo 1 > /proc/sys/net/ipv4/ip_forward"
sudo iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
sudo sh -c "iptables-save > /etc/iptables.ipv4.nat"
sudo nano /etc/rc.local
iptables-restore < /etc/iptables.ipv4.nat
sudo systemctl unmask hostapd
sudo systemctl enable hostapd
sudo systemctl enable dnsmasq
sudo systemctl start hostapd
sudo systemctl start dnsmasq

```

FUF1, 2025-11-20:
Thank you @nickpourazima for the guidance! And sorry for the late reply. After wedding, moving and not having a good "infrastructure" I managed to organize a suitable display and a second SD-card and it works flawlessly. I highly recommend this awesome modification with the AP. It makes it so nice to use and a nice addition to make it USB-C powered and therefore just plug-and-play everywhere. As far as I understand, the only real line of defence are the AP credentials. Does it break the system if change the username and password for the system via the raspi CLI?

[image]

sascha-schieferdecker, 2025-12-09:
I think something like this could be a viable solution:

https://github.com/gitbls/autoAP

It would be best if the switch happens automatically so you can always connect to the Pi.

sascha-schieferdecker, 2025-12-19:
Yes, that is a problem with the RPi imager. But they have fixed in 2.0.2. It does not seem to be linked on the official homepage yet:

https://github.com/raspberrypi/rpi-imager/releases/tag/v2.0.2

I'll be looking into this feature in the next days.