nickpourazima/rotary-phone-audio-guestbook #81: Sourcing Lav Mic

jacob-r-g, 2025-04-30:
Hello! Thank you Nick for this amazing project. I followed it for our upcoming wedding, got it workign quick and love it! However, the sound quality is kind of trash (as you point out in your documentation on adding a lav mic). I have tried to find a lav mic to follow your instructions for replacement, but after buying 2, found they are both 2-wire type mics.

**Do you have any tips for sourcing a 3 wire Lav mic that could be used according to your guide?**

nickpourazima, 2025-04-30:
Hey @jacob-r-g, thanks for the support! You can actually use the two wire lav no problem, that's what I ended up using as well. The "3rd" wire is the copper shielding which you just have to peel away and twist together to act as the ground wire. Let me know if you need any more help, send me a picture if it's easier - best of luck!

jacob-r-g, 2025-04-30:
Thanks! I shared a picture here. can you let me know if you think these mics would work? If so,  i'll try to figure out the wiring

[image]

nickpourazima, 2025-04-30:
Oh wow these actually look like pretty fancy lav mics! You're spot on with your designation - the red labelled should be the signal and if you can bundle enough of the other yellow labelled signal for GND you should be good. However, just be careful when soldering because those wires typically have this enamel coating on them. I've burned it off in the past before I've soldered but check YouTube for some tips. Here's a brief guide:
https://electronics.stackexchange.com/questions/60031/what-material-are-the-thin-colored-wires-in-my-headphone-cable

jacob-r-g, 2025-05-07:
thanks @nickpourazima! Im not sure how fancy they are...they were $6 and $15 on Amazon, but I guess they could be fancy! So is it correct that the red signal wire from the lav would go to the red+ wire from the phone jack (fromerly plugged into the carbon mic, the ground wire from the lav would tap into the ground (green) cable attached to the headset speaker and the black (M-) formerly going to the carbon mic would just be loose and floating?

jacob-r-g, 2025-05-11:
I was unable to get it working by wiring up any of my Lav mics, but i found an alternative solution to improve the sound quality, by using a drop in replacement [electret condesner micropophone](https://payphone.com/EMS-94-Electret-Condenser-Microphone.html)

nickpourazima, 2025-05-12:
Oh cool! Sorry for the late response things have been busy for me these past few weeks. Let me know if I can be of any further help!

closed 2025-08-03 by nickpourazima