nickpourazima/rotary-phone-audio-guestbook #89: Hook handling is not stable

sascha-schieferdecker, 2025-08-19:
First: Thanks a lot for your project 👍 

I am trying to use it for a birthday party. I got everything working hardware wise, but the handling of the hook is not stable in my setup. It works okay first time, but if I put down the hook if the greeting is playing, another pickup won't be registered. It seems to me that the threads are not terminated correctly. Do you have an idea? I already played with bounce times, but that did not have any effect. Do you have an idea where the problem could stem from?

BR and thanks

sascha-schieferdecker, 2025-08-19:
This simplified code works without problems in my setup, maybe it helps. I really suspect the thread handling, but python is not my home turf. I had an arecord process running even way before recording and even if I disabled beep recording.

```python
#!/usr/bin/env python3
import RPi.GPIO as GPIO
import subprocess
import time
from datetime import datetime
from pathlib import Path
import os

# GPIO configuration
PIN_HOOK = 22   # Handset hook (off-hook = LOW)
PIN_SHUTDOWN = 24  # Shutdown button (hold LOW ≥4s)

# Paths
BASE_DIR = Path(__file__).resolve().parent
SOUNDS_DIR = BASE_DIR / "sounds"
GREETING_FILE = SOUNDS_DIR / "greeting.wav"
BEEP_FILE = SOUNDS_DIR / "beep.wav"
RECORDINGS_DIR = BASE_DIR / "recordings"
RECORDINGS_DIR.mkdir(exist_ok=True)

recording_proc = None

def play_wav_interruptible(file_path: Path) -> bool:
    """Play a wav file, but stop if hook goes HIGH (on-hook)."""
    if not file_path.exists():
        print(f"ERROR: Missing audio file: {file_path}")
        return False

    print(f"Playing: {file_path}")
    proc = subprocess.Popen(
        ["aplay", "-q", "-D", "plughw:1,0", str(file_path)]
    )

    try:
        while proc.poll() is None:
            if GPIO.input(PIN_HOOK) == GPIO.HIGH:
                print(f"Interrupted {file_path} (on-hook)")
                proc.terminate()
                try:
                    proc.wait(timeout=1)
                except subprocess.TimeoutExpired:
                    proc.kill()
                return False
            time.sleep(0.05)
    except Exception as e:
        print(f"Playback error: {e}")
        return False

    return True

def start_recording() -> subprocess.Popen:
    """Start arecord process (CD quality)."""
    timestamp = datetime.now().strftime("%Y%m%d-%H%M%S")
    out_file = RECORDINGS_DIR / f"{timestamp}.wav"
    print(f"Recording to {out_file}")
    proc = subprocess.Popen(
        ["arecord", "-q", "-f", "cd", "-t", "wav", "-D", "plughw:1,0", str(out_file)]
    )
    return proc

def stop_recording(proc: subprocess.Popen):
    """Stop an arecord process if running."""
    if proc and proc.poll() is None:
        print("Stopping recording...")
        proc.terminate()
        try:
            proc.wait(timeout=2)
        except subprocess.TimeoutExpired:
            proc.kill()

def check_shutdown_button(hold_time=4.0):
    """
    Monitor shutdown pin. If held LOW ≥ hold_time seconds, shut down system.
    """
    if GPIO.input(PIN_SHUTDOWN) == GPIO.LOW:
        start = time.time()
        while GPIO.input(PIN_SHUTDOWN) == GPIO.LOW:
            if time.time() - start >= hold_time:
                print("Shutdown button held LOW for 4s -> shutting down...")
                # Make sure recording stops before shutting down
                stop_recording(recording_proc)
                os.system("sudo shutdown -h now")
                return
            time.sleep(0.1)

def main():
    global recording_proc

    GPIO.setmode(GPIO.BCM)
    GPIO.setup(PIN_HOOK, GPIO.IN, pull_up_down=GPIO.PUD_UP)
    GPIO.setup(PIN_SHUTDOWN, GPIO.IN, pull_up_down=GPIO.PUD_UP)

    print("Waiting for handset (GPIO22 LOW = off-hook). Press Ctrl+C to exit.")

    prev_state = GPIO.input(PIN_HOOK)

    try:
        while True:
            state = GPIO.input(PIN_HOOK)

            # Off-hook (HIGH -> LOW)
            if prev_state == GPIO.HIGH and state == GPIO.LOW:
                print("Handset lifted (off-hook)")

                print("Waiting 1 second before greeting...")
                for _ in range(10):
                    if GPIO.input(PIN_HOOK) == GPIO.HIGH:
                        print("Handset replaced during silence — aborting.")
                        break
                    time.sleep(0.1)
                else:
                    if not play_wav_interruptible(GREETING_FILE):
                        prev_state = GPIO.input(PIN_HOOK)
                        continue
                    if not play_wav_interruptible(BEEP_FILE):
                        prev_state = GPIO.input(PIN_HOOK)
                        continue
                    if GPIO.input(PIN_HOOK) == GPIO.LOW:
                        recording_proc = start_recording()

            # On-hook (LOW -> HIGH)
            if prev_state == GPIO.LOW and state == GPIO.HIGH:
                print("Handset replaced (on-hook)")
                stop_recording(recording_proc)
                recording_proc = None

            prev_state = state

            # Always check shutdown button
            check_shutdown_button(hold_time=4.0)

            time.sleep(0.05)

    except KeyboardInterrupt:
        print("\nExiting...")
    finally:
        stop_recording(recording_proc)
        GPIO.cleanup()

if __name__ == "__main__":
    main()

nickpourazima, 2025-08-22:
@sascha-schieferdecker @CringeWarframePlayer 

I’ve pushed a new branch with potential fixes for the hook-handling issue:  

https://github.com/nickpourazima/rotary-phone-audio-guestbook/tree/89-hook-handling-is-not-stable

Can you try pulling it down and running again?

```bash
git fetch origin
git checkout 89-hook-handling-is-not-stable
```
I’ve refactored the hook logic and improved interrupt handling to prevent stray `arecord`/`aplay` processes. Hopefully this resolves the instability you’ve been seeing.

nickpourazima, 2025-09-03:
Nice work @calombus101! This looks like a solid fix for the hook debouncing issues. A few suggested improvements to make it even more robust:

1. Add a cleanup method to ensure all session flags and threads reset properly:

```python
def cleanup_session(self):
    """Ensure all session-related flags and threads are properly reset"""
    self.current_event = CurrentEvent.NONE
    self.recording_watchdog_active = False
    if hasattr(self, 'timer') and self.timer:
        self.timer.cancel()
```

2. Clarify watchdog state tracking - The `self.watchdog_active` flag seems to only track the recording watchdog, maybe rename it to `self.recording_watchdog_active` for clarity.

3. Consider adding a small delay between session attempts to prevent thread proliferation during rapid toggling.

Would you be willing to submit this as a PR? 

It would help a lot of people having similar issues, and having it in the main repo would make it easier for everyone to benefit from your work. Thanks for sharing this fix!

calombus101, 2025-09-03:
Thanks! @nickpourazima 

Happy for any of these fixes and code to be adapted, fixed, added to the main project. I'm just grateful to all the work done by you and others before me.

So yes it can be submitted to a pull request. But if you can easily do that then please do.. as I dont know how to yet. (Literally created a Git account just to post my fix on this)

1 question about the cleanup_session(self)

When / at what points should i be calling this function in the code?

nickpourazima, 2025-09-03:
No worries! Thanks for being willing to contribute.

### For `cleanup_session()`:

Call it at the end of any recording session. 

In your code, add it:

- At the end of `on_hook()` after setting current_event to NONE
- At the end of `time_exceeded()` after stopping recording
- At the end of `released_record_greeting()`

Basically anywhere you're ending a session and resetting to idle state.

### For the PR:
Since you're new to GitHub, this is actually a great learning opportunity! GitHub has a good guide here: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

### Basic steps:

- Fork the repo (button on top right)
- Make your changes in your fork
- Click "Contribute" → "Open Pull Request"

Give it a shot - the community can help if you get stuck. If you really can't figure it out, maybe someone else can shepherd your code into a PR. Check the discussions/issues - others might be willing to help.

Good luck! Your fix will definitely help a lot of people.

sascha-schieferdecker, 2025-12-07:
I've provided a pull request that is rather a rewrite of the hook handling, because it is a simple polling loop. I could not find any faults with it, maybe it helps others. I removed unnecessary config options that are not required. I also removed beep_include_in_message because I do not see any real use in that feature, hope that is okay.